iNavFlight/inav #1: Flight - New navigation stack

digitalentity, 2015-05-21:
This is to track and discuss progress of nav-rewrite branch

marksev1, 2015-06-09:
Congrats on your latest test flight, looks promising :)

marksev1, 2015-06-11:
I'm liking the progress, hopefully the crash during rth didn't do too much damage.

digitalentity, 2015-06-12:
Not much - broke an arm and a prop, teared off a wire from one motor. Lots of crashes during development of completely new code was not unforseen and I've got alot of spare parts, fortunately they are very cheap on eBay. I think I've found an issue with RTH, wil try and test again in a few days. Currently the thing that bothers me most is not working manual control during PH (#11) as I have no idea why this is happening...

marksev1, 2015-06-13:
Are nav pids transferrable from harakiri?

digitalentity, 2015-06-13:
Not really, PIDs behave differently than Harakiri's.

marksev1, 2015-06-21:
DE do you have plans for also improving bmp085 althold. I recall harakiri has done some improvements for that baro, but when I was glancing that code I couldn't understand a lot.. :)

digitalentity, 2015-06-21:
Probably yes, but currently I'm focused on getting navigation working. When it will be mostly flyable we can think of improving things.

digitalentity, 2015-06-23:
@marksev1, I've looked at Harakiri code for baro, it occurs to me that we've recently implemented the similar filtering in 6e041c57dd00f6e8ec195dc87e3402d9bd8a34c9.

marksev1, 2015-07-18:
Digitalentity, what the latest status on ph, rth and ah?

digitalentity, 2015-07-18:
@marksev1 
AH and PH work to certain level - require tuning, but algorythms are stable enough. RTH is not so good. Flight test of RTH today was a complere disaster. Nothing related to RTH seem to work. But that's fine - I can recall times when even AH wasn't working.

strips, 2015-08-12:
Awesome work! I really hope you can make this work so it will be reliable!

digitalentity, 2015-08-13:
@strips 
Thanks! Code is getting more stable every day, improvements are made all the time. I hope I'll get this code past "flyable" milestone in about a month.

digitalentity, 2015-08-28:
Some new changes are ready to be flight-tested. New binary **cleanflight_NAZE.navigation-rewrite.Aug28.hex** is built from 25a0e4f0e2cb47ad1729e835c476ce6633775e34 and is available in my [Dropbox](https://www.dropbox.com/sh/4r5374mos7471q2/AACppn0Mphw45p-4TJ7lZLlOa?dl=0).
Most significant changes:
* code refactoring (slight possibility I might have broken some functionality)
* re-added sonar support for low altitude ALTHOLD, needs careful testing
* added compensation for BARO drift from GPS, needs testing - enable ALTHOLD and hover for couple of minutes - shouldn't drift for more than 0.5-1 meter
* PH changes, should tolerate wind better
* RTH changes, verify that PH is working before testing this
* reworked landing code, should now descend faster.

digitalentity, 2015-08-29:
Bugfix build (commit da8751c9f52c4e5e92574a08924ad0d8bb061c58). New hex file: **cleanflight_NAZE.navigation-rewrite.Aug29.hex**
Changes:
* Baro offset correction from GPS removed - GPS altitude turned out to be even less reliable than baro. Using GPS altitude will only be available for fixed-wing if no other source of altitude is available.

digitalentity, 2015-09-05:
New builds: **cleanflight_CC3D.navigation-rewrite.Sep06**
Changes:
* Improvements for XY-controller code
* New set of default settings. Looks like #4 is going to be closed soon.

digitalentity, 2015-09-06:
Posted a thread on RCGroups for discussion: http://www.rcgroups.com/forums/showthread.php?t=2495732

digitalentity, 2015-09-08:
New build: **Sep08**
Changes:
* Updated to latest master (including new failsafe functionality)
* Improved ALTHOLD and POSHOLD (GPS_CRUISE) mode, should now stop faster when pilot releases the stick
* Added a filter to GPS coordinates, should reduce GPS noise

digitalentity, 2015-09-11:
Build **0910** uploaded. 
Updated to include changes from latest master (including full support for external baro/mag).
Some CLI parameter names changed, new navigation-related CLI parameters added. This build includes stronnag's implementation of LTM. support for dodo baro sensor.

digitalentity, 2015-09-14:
Just uploaded a **0914** build.
Changes:
* (tested) GPS update rates of >2Hz are now supported
* (tested) Limited pilot's ability to control altitude during RTH only to HEAD_HOME phase. This fixed a bug that prevents autolanding if throttle stick > 50% when landing
* (untested) GPS loss during PH/RTH should be now handled properly - force ANGLE mode and wait for GPS to restore. Allow pilot to control the copter pitch/roll directly. In future this will change for RTH mode and will force autolanding

This build also includes support for LTM telemetry and BMP280 baro.

digitalentity, 2015-10-07:
Uploaded a 1007 build
Changes:

* Rebased to latest "master", support for BMP280 and Naze rev.6 out of the box
* (partially tested) Model-based inertial position estimator (#23) for PH/RTH
* (partially tested) Dead reckoning possibility (poshold without GPS)
* (untested) Improvements for autonomous navigation code (PH/RTH)

bk79, 2015-10-07:
@digitalentity A couple of question:  Did you separate DR from PH? Did you fix the throttle manual control when in PH/DR? DId you try PH/DR when in acro mode?

digitalentity, 2015-10-07:
@bk79 
Not yet, GPS-based code is a higher priority at the moment.
Currently dead-reckoning is enabled via CLI and its parameters have been changed to slow down drift. 
Poshold does not control throttle at all. You have to manually maintail throttle or use althold. 
When PH is activated firware automatically enables ANGLE mode, there is nothing like PH in acro :)

bk79, 2015-10-08:
@digitalentity no what I mean is that if I'm in acro and I activate PH, regardless my throttle stick position it goes down

digitalentity, 2015-10-08:
@bk79 
Ok, I see. will test and fix this.

digitalentity, 2015-10-12:
New build **1010**. Built from c0d3edd40ed7e3cbc1816c91627b697e7c5d005d
Changes:
* Flight mode IDs changed for compatibility with mw-osd. Old dumps will not restore modes (aux) correctly.
* Dead reckoning improvements
* Automatic magnetic declination (experimental)
* Less agressive configuration defaults
* Bugfix - home position was not updated
* Bugfix - throttle control in POSHOLD without ALTHOLD
* Bugfix - consistent manual control in POSHOLD and ANGLE modes
* Code cleanups

digitalentity, 2015-12-20:
Further news in RCG thread http://www.rcgroups.com/forums/showthread.php?t=2495732

closed 2015-12-20 by digitalentity