iNavFlight/inav #3: Flight - NAV_POSHOLD erratic behaviour

digitalentity, 2015-05-22:
During flight testing today I switched to NAV_POSHOLD mode and copter began to change pitch/roll randomly. Possible causes: 
- wrong alignment or flawed calibration of magnetometer
- code error
- unhealthy default PIDs

digitalentity, 2015-05-22:
Found a possible cause - both PITCH and ROLL axis were using the same PID regulator (I and D accumulators were shared). Fixed in effe239d17969e7dc3e3ddf40ee00b8c37145e20, might fix the error, needs testing.

digitalentity, 2015-05-28:
Another bug -  X and Y velocities from IMU are rotated compared to LAT and LON velocities from GPS. When aircraft is heading north this does not affect calculations, but when heading != 0 everything goes wrong. Fix on the way.

digitalentity, 2015-06-01:
It just occured to me that IMU calculates accelerations in NED coordinates, X axis pointing north and Y - east. I was doing all my calculations for nav-rewrite in wrong coordinates - ENU (X pointing EAST, Y - north)! That's why I was unable to get PH working and crashed into a tree yesterday! I feel a total dumbhead!

marksev1, 2015-06-01:
So when is next test flight planned? :-)

digitalentity, 2015-06-01:
Next testflight would be when I get a new GPS module instead of destroyed one. Hopefully it won't take too long... :-)

marksev1, 2015-06-02:
Maybe some sort of environment for testing/simulation (hardware in the loop) "in silicio" could help you in your testing, maybe that qgroundcontrol, treymarc ported msp to it https://github.com/treymarc/multiwii-mavlink-gc
See video here where he tested PatrikE's baseflight for plane stuff: https://vimeo.com/92207766

digitalentity, 2015-06-02:
Cool! I'll look into it while I'm waiting for GPS module

marksev1, 2015-06-05:
Be sure to make a video of the simulation, if it looks good :D

digitalentity, 2015-06-05:
That would be a "must do". So far I was not able to set HIL simulation, probably need to patch cleanflight firmware to make this work. I was also busy improving ALTHOLD - now it does throttle tilt compensation with completely automatic tuning. Basically, after a minute of slow flight and hovering with ALTHOLD enabled, you can do some rapid flying without touching throttle stick and without loosing altitude.

digitalentity, 2015-06-06:
Got a spare GPS module from a local rc model workshop, flight tested some latest bugfixes. 7160efe01cf33d061faf18614ea613c376d8baa5 finally got basic poshold working (https://youtu.be/QSGkQENC3ek). There are some bugs though: #6, #7, #8, #9. As basic POSHOLD functionality is now working, this issue is resolved.

closed 2015-06-06 by digitalentity