iNavFlight/inav #10: Flight - RTH does not work

digitalentity, 2015-06-11:
Tried to enable RTH today - resulted in immediate crash. Motors simply stopped. Need to investigate.

digitalentity, 2015-06-11:
219c21ee52ab5b334378c1ff0aa775cee6b114e3 is a possible for for instant disarm instead of graceful landing.

digitalentity, 2015-06-13:
Autolanging seem to work now, however `isLandingDetected` does not work properly. RTH itself is still untested.

digitalentity, 2015-06-19:
Some fixes to autolanding - ignored throttle input from pilot during autoland phase. It should LAND regardless of throttle stick position. If pilot desires otherwise - disable RTH and take over.

dkisselev, 2015-06-19:
Just to put this out there - you're not just commenting into the void, I (and I'm sure a few others) are following and paying attention to your progress. I'm really looking forward to trying things out when they approach stability. 

Feel free to ping me if you need a second tester for anything. (I might take a few days to get back to you because I need to resolder my GPS, but I'll help out wherever I can)

digitalentity, 2015-06-20:
Good to hear! I've noticed a few watchers to this repo, some also comment on my new code.

You can grab the latest code from navigation-rewrite branch and give it a try. ALTHOLD and POSHOLD are mostly flyable, but please avoid RTH for now - code is there, but it is buggy and mostly untested. I am very interested in testing new code on 250 size copters as I don't have one. Suggestions on default PID's that would suite most users are also very appreciated :-)

skaman82, 2015-06-28:
I am following your work for a while now. You making great progress! I would like to test your code on my 250 quad. Just need a hex file for the naze32. Where can I get the latest build?

digitalentity, 2015-06-29:
@skaman82 thanks! I started placing my dev builds here: https://www.dropbox.com/sh/4r5374mos7471q2/AACppn0Mphw45p-4TJ7lZLlOa?dl=0
**cleanflight_NAZE.nav-rewrite.Jun10.hex** is flight tested, bugs are known and probably fixed in **cleanflight_NAZE.nav-rewrite.Jun28.hex**.
The latter is NOT flight tested and has some new code. It should be flyable.
Please be advised, that default PIDs might not work for you, you'll have to find your values by trial and error.
Currently you can try new ALTHOLD and POSHOLD mode. ALTHOLD requires barometer, POSHOLD required gps & compass. You can also play with `nav_gps_cf`, `nav_expo`, `nav_lpf`, `nav_use_midrc_for_althold`, `nav_throttle_tilt_comp` and `nav_lock_until_takeoff` cli variables.
`nav_gps_cf` - the LOWER the value the closer estimated speed is to gps speed (dont go to high here)
`nav_expo` - 1 - 99 % defines the actual expo-curve applied for GPS
`nav_lpf` - low pass filter cut frequency for navigation PID D-term calculation
`nav_use_midrc_for_althold` - don't remember throttle value when althold was initiated, assume that throttle stick is at middle equals zero climb rate (may cause drop in altitude when swithing to ALTHOLD)
`nav_throttle_tilt_comp` - calculate and use automatic throttle tilt compensation
`nav_lock_until_takeoff` - navigation modes won't enhage until throttle is raised over 50% ot least once after arming.

dkisselev, 2015-06-29:
@digitalentity what kind of crash was it? Is it something that you could recover from if you see it going sour and flip back to horz while in testing?

digitalentity, 2015-06-29:
@dkisselev that was a false landing detection during descent that disarmed my copter 5m above ground. For now I've disabled automatic disarm until a reliable landing detection gets developed #21. Other possible bugs in RTH code *should* be manually recoverable.

digitalentity, 2015-08-11:
Automatic landing code tested today Looks like it's working, but copter is rotating while descending - probably something is wrong with heading hold during RTH.

digitalentity, 2015-08-13:
RTH is now working as expected - climb to 10m baro altitude, head home, land. Landing detestion and autodisarm not working yet.

closed 2015-08-13 by digitalentity

dkisselev, 2015-08-13:
Amazing. Thanks Konstantin!

digitalentity, 2015-08-13:
As an improvement I am planning to add 4 modes of RTH:
1) Keep current altitude, head straight home
2) Climb to a predefined configurable altitude above launch point, head home
3) Track the highest altitude during whole flight, climb there, head home
4) Climb to current altitude + some configurable extra, head home
What do you think?

dkisselev, 2015-08-13:
I like it. But I don't think they should be 4 unique modes. Instead "altitude" should just be a setting inside of the GPS section of the configurator.

Perhaps with a knob/switch adjustment that lets you adjust what the configured altitude/offset should be while the craft is in the air.

digitalentity, 2015-08-13:
No, there would still be only one flight mode - RTH. It's behaviour thats what would be configurable.

digitalentity, 2015-08-13:
Please see the video of today's RTH test - https://youtu.be/f2NUYgHNiQI

Andycb1, 2015-09-03:
As I fly in several different areas I like the idea of being able change the RTH altitude mid-flight rather than always having it set in the configurator. So it could be set up so that if you don't have a spare AUX channel you set an altitude in the configurator but I'd like to be able to set a pot on my controller to set the altitude.

So if the min and max values of my pot are 1000 and 2000, I could set 1000 = 5metres and 2000 = 15m so if on my controller it's currently at 1750, my quad would return at an altitude of 12.5m.

I think it would be a nice feature to have but that is an extra to add once the main code is working.

And with mode 4 of RTH, is it saying that if you're at 5m when you activate RTH and you have a buffer of 2m, the quad would move to 7m altitude before returning home?

digitalentity, 2015-09-03:
@Andycb1 that would be an extra feature to consider after we get the main code working.

Mode 4 of RTH is AT_LEAST_ALT, in your situation you'd be returning at 5m (5m > 2m)
In mode 1 of RTH (EXTRA_ALT), you'd be returning at 7m (5m + 2m)