iNavFlight/inav #11: Flight - Using sticks in POSHOLD mode result in failure to hold position

digitalentity, 2015-06-11:
When using sticks to adjust position (mostly yaw stick) copter responds, but after that it fails to maintain position and begins to fly in some direction. Disabling and re-enabling PH stops the drift.

digitalentity, 2015-06-11:
219c21ee52ab5b334378c1ff0aa775cee6b114e3 is a possible fix for this. Should confirm that issue persists and if so, record some blackbox logs to investigate further.

digitalentity, 2015-06-12:
I've decided to create two variants of processing user-control in PH mode: NAV_GPS_ATTI and NAV_GPS_CRUISE (named per NAZA's GPS-Atti and GPS-Cruise modes) - 714b3162ba0f768c6735670ff76a3ab1ae897f9d.
In ATTI mode pilot will control attitude directly, while in CRUISE mode pilot will control general velocity and let the copter figure attitude for itself. I think the current problem might be related to manual velocity control and thus will be limited to CRUISE mode only. In a few days I'll try to flight-test this.

digitalentity, 2015-06-13:
This is now limited to GPS_CRUISE mode. So far I'll leave GPS_CRUISE mode alone and stick with GPS_ATTI. After I get everything else workins, I'll get back to this.

csurf, 2015-08-19:
what's the status on these flight mode changes? 
Is GPS ATTI working? 
Is it a preferable alternative flight mode to the traditional 'GPS HOLD+BARO' mode? 
Will GPS ATTI automatically hold altitude when moving the sticks and flying around (assuming i have a baro sensor)?
Does BARO need to be enabled, or is it enabled by default?

digitalentity, 2015-08-20:
@csurf 
User control mode defined a way how pilot's input is treated when in POSHOLD mode. It is controlled by `nav_user_control_mode` variable.
1) GPS_ATTI is a default mode (nav_user_control_mode=0), in this mode user controls angle, much like in ANGLE mode. When pitch/roll stick is moved, XY-position controller is bypassed and stick input is passed directly to ANGLE-mode controller. After stick is returned to center, new hold position is recorded and used.
2) GPS_CRUISE (nav_user_control_mode=1) is different. In this mode user controls velocity and XY-controller is always active. This mode was changed recently, but I didn't have a chance to test it yet.
ALTHOLD is completely different and will be always active if enabled.

digitalentity, 2015-09-05:
Flight tested today. Working as expected.

closed 2015-09-05 by digitalentity