iNavFlight/inav #13: unneccessary calculations

bubi-007, 2015-06-19:
https://github.com/digitalentity/cleanflight/commit/6adef51948f198c662cb42c9c49406622529455f

the timeslice is valid for all axis.
there is no need to calculate all individual time slices.

digitalentity, 2015-06-19:
What timeslice are you referring to? `accTimeSum` is calculated individually for all axis because reference velocity sampling rate is differect at least for X/Y and Z axis.

digitalentity, 2015-06-19:
`imuSampleAverageAccelerationAndVelocity` is called independantly for each axis when sampling is needed and when reference velocity is ready to be updated using `imuApplyFilterToActualVelocity`. This in general occurs at different intervals for X/Y axis (using GPS as a reference) and Z (using sonar/baro as a reference).

closed 2015-06-20 by digitalentity