iNavFlight/inav #14: NAV - Use raw GPS velocity in calculations

digitalentity, 2015-06-20:
Current code calculates GPS velocity from coordinate difference. @bubi-007 suggested that raw velocity from GPS unit should be used instead. I think it's worth trying out.

digitalentity, 2015-06-20:
This can be done by using `GPS_speed` and `GPS_ground_course`, but code MUST check that GPS unit provide these values - it is not a GGA message, it's a RMC message and it can be disabled in GPS module for some reason.

digitalentity, 2015-06-20:
GPS data validity addressed in 6375c485a53eea080cd23bf85c0b0b3a75548305

bubi-007, 2015-06-20:
you 'can' disable everything.
but you can subscribe for the velocities in the ned frame:
0xB5,0x62,            // Header             Turn On UBLOX NAV-VELNED Msg
0x06,0x01,            // ID
0x03,0x00,            // Length
0x01,                     // Msg Class    NAV
0x12,                     // Msg ID       VELNED
 0x01,                    // Rate         1
0x1E,0x67,            // CK_A, CK_B

depends on what you like.

digitalentity, 2015-06-21:
The navigation code should not depend on a specific hardware feature (in your example an UBLOX GPS module). If GPS for whatever reason does not provide velocity information, navigation code must still function properly.

bubi-007, 2015-06-21:
i just take what works - ublox does and the rest is not so reliable.

digitalentity, 2015-06-21:
Sticking with ublox-specific features would effectively remove support for nmea-only gps receivers. We can detect provided information and use what's best - that would be velocity info if available and coordinate difference as a fallback. This will improve things for ublox users and will not break anything for others.

digitalentity, 2015-06-27:
Done in 05c115ba6edb79f5f64f8c756c2787d14e1336d0. Untested.

digitalentity, 2015-07-05:
Did a test today. Didn't work as good as I thought it will be. However the test was done on poorly tuned copter, so it could be vibration messing up with velocity calculation from accelerometer. Need to re-tune and re-test. If it will work less reliable than coordinate derivative - this will be removed.

digitalentity, 2015-07-17:
Did another test today. Was working MUCH worse than plain old coordinate derivative. Closing.

closed 2015-07-17 by digitalentity

bubi-007, 2015-07-18:
the current approach sucks.
bf and cf need a complete rewrite on this subject.

digitalentity, 2015-07-18:
@bubi-007 
Please assist in getting this working,

bubi-007, 2015-07-18:
first i am integrating altitude hold using gps and accelerometer.
then i go into navigation.

digitalentity, 2015-07-18:
@bubi-007 
Can you please share your code?