iNavFlight/inav #20: GPS - Modify GPS so it provides MSL altitude in cm

digitalentity, 2015-06-29:
Current GPS code provides altitude in meters, insufficient for good altitude hold, NMEA is capable of 10cm resolution, should use all available precission.

csurf, 2015-08-19:
@digitalentity, I'm confused, is the CF code currently using cm, 10*cm, or meters in its calculations?

digitalentity, 2015-08-19:
@csurf, I was confised as well. Most of the code uses cm, however gps.c sets GPS_altitude to meters. Similar thing for angles: imu calculated RPY angles in 0.1 degrees, however legacy navigation code (and my code as well) uses 0.01 degrees. I might think about fixing that in future.

csurf, 2015-08-19:
so, will there be any problems with the recent issue/code change with defining NAV_RTH_ALTITUDE to use cm? I'm still not sure what type of units the nav code is using, need to know this before I try to flight test. please advise...

digitalentity, 2015-08-19:
I designed nav-rewrite code is using cm for distances everywhere, except for `distanceToHome` variable (calculated in meters for compatibility with the rest of the code)

digitalentity, 2015-08-28:
I figure there is already an issue for GPS altitude opened in cleanflight: https://github.com/cleanflight/cleanflight/issues/367

digitalentity, 2016-01-23:
Done.

closed 2016-01-23 by digitalentity