iNavFlight/inav #21: NAV - Landing detection

digitalentity, 2015-06-29:
Ideas, anyone?

bk79, 2015-06-30:
sonar should help a lot, but if we want to make it without sonar, we can set a fixed descent speed setting, and when we get with speed below a certain threshold we consider it landed and we lower motor again, another check on the vertical speed if no vertical speed is detected in 10 second lowering the motor we can disarm

bk79, 2015-06-30:
another option could be to "train" the landing speed. In altitude hold we have and average value of command, if we can record during hoovering just before landing using PI with tight clamp on low and high level would let it descend slowly. it's not a perfect way but it can be a start in order to have it working

digitalentity, 2015-07-01:
@bk79 
I like the idea with vertical speed. Current code calculated average vertical velocity every 250ms. If this average velocity is low enough for 5-10seconds and throttle is low enough, we can assume landing.

bk79, 2015-07-01:
how are you going to measure the descend speed? using still GPS? I think that integrating Z axis acceleration would be reliable

digitalentity, 2015-07-01:
Current code uses complementary filter to blend accelerometer integration result and altitude derivative from barometer.

bk79, 2015-07-01:
@digitalentity  nice, also sonar would help specially when below 2 m that is actually the last part of descend...a good part would be if (sonar is present and measure is valid and estimated height from baro is < 3 m then use baro derivate for both descent speed and positioning

digitalentity, 2015-07-01:
@bk79 
Code that would use sonar for low altitude positioning is already there, but untested yet.
My support for i2c_nav does not include sonar data, only raw gps data is being read and used. Adding support for sonar is a good idea. I'll get to it after nav-rewrite will get to stable state. I've opened an issue for it (#22) so it won't be forgotten.

bk79, 2015-07-20:
@digitalentity  I was wondering if we cannot trust speed calc for now why don't we just perform steps lowering: ie copter is at 400 cm in althold, landing is requested , althold request will ask 375 cm, once there and stable for more than 3-4 s, request  will go down to 350 when hoovering  at 350 setpoint down again 25 cm and so on until the so if far below the request

digitalentity, 2015-07-20:
Hm... Actually this might work. Don't trust speed, trust altitude. If whan landing we request descent rate of 25 cm/s and don't see altitude 2.5m altitude drop in 10 seconds, assume that we've hit the ground.

bk79, 2015-07-20:
Actually the same solution can be used for rth and wp nav, consider that distance from home is far than 3m create an intermediate wp as new setpoint

digitalentity, 2015-08-12:
@bk79
Latest tests showed that I can't trust baro when close to the ground due to ground effect. Probably the most reliable way to detect a touchdown is detecting a rapid growth in accelerometer readings. Take a derivative of an accelerometer and look for a large positive spike. Having a sonar will help as well.

bk79, 2015-08-12:
Good idea baro is reliable over 1.5 m with very little wind and can only be
used for alt hold, shielding is necessary even. Sonar is perfect for
landing even if it suffer from the surface type on grass it becomes little
noisy. But with the smoothing function I've done my jumps in readings
become very little and it can be considered thrustwhorty specially for
landing

digitalentity, 2015-11-27:
If sonar is present it will be used to detect landing - 6afd3741b23989d6698f1d627693b710a7acac4c

digitalentity, 2015-12-05:
Appears that landing detection works, but it is dependent on having a sane value of nav_vel_i parameter. This issue is now PID related.

digitalentity, 2015-12-20:
Lack of landing detection does not make the code unflyable.

commit 9c5794bd (2016-04-08): NAV: Disarm after landing. Controlled by 'nav_disarm_on_landing' setting. Closes #21
closed 2016-04-08 by digitalentity