iNavFlight/inav #22: I2C_NAV - Add support for sonar altitude

digitalentity, 2015-07-01:
Generic i2c_nav board has pins for connecting a sonar. There should be a possibility to read from it.

bk79, 2015-07-01:
I've ordered an I2c_nav board yesterday, as soon as I receive it I'll fork and work by myself to give help, I'm also planning to add maplemini support that is still waiting to get merge from main stream into this.

digitalentity, 2015-07-01:
Excelent! Eventually I hope to get navigation-rewrite and nav-i2c-gps merged into official cleanflight for everybody's benefit. When forking this repo be aware that *master* branch is frequently synced to official cleanflight's master. Don't post PR's for merging to *master* - they will not get merged.

bk79, 2015-07-07:
@digitalentity Why do we have the need of using 32bit to get sonar evaluation if 300 cm is more than too much for valid reading?

digitalentity, 2015-07-07:
@bk79 Yes, int16_t should be more than enough, but I would rather keep navigation-rewrite branch free from hardware/sensor/driver-related changes (besides changes that are needed for navigation to work). Improvements and optimizations can wait. You can also open an issue for this on official cleanflight repo.

bk79, 2015-07-08:
That would simplify i2c sonar if you look at i2c-nav code from eosband
sonar register sends one uint16 and since right now I'm trying to figure
out where is the best place where I can put the code: modifying gpsmsg
struct adding one more field to read during i2c GPS reading and then
casting at sonar.c the already taken result; or calculating everything
inside sonar.c  what do you think?

digitalentity, 2015-07-08:
Hmm... I would prefer not to mix things up - add i2c-sonar driver that would mostly duplicate i2c-gps driver and periodically read sonar altitude from i2c-nav board. This way we can add support for sonars with native i2c (i.e. Devantech SRF02 http://www.robotshop.com/en/srf02-ultrasonic-range-finder.html)

bk79, 2015-07-08:
@digitalentity I thought different way to put it in, but the only easy way since this is something specific of the i2c-nav board is to set it inside of the i2c-nav gps, adding one more flag to the gpsmsg and one more line for the sonar code. one more request won't change the speed and the same time if the sonar is installled on the board the rest of the code will receive a -1 that means no sonar valid value:

something like 

            i2cRead(I2C_GPS_ADDRESS, I2C_GPS_ALTITUDE,      2, (uint8_t*)&gpsMsg->altitude);
			i2cRead(I2C_GPS_ADDRESS, I2C_GPS_SONAR_DISTANCE,2, (uint8_t*)&gpsMsg->sonar_distance);    			
            gpsMsg->hdop = 0;
            if (gpsMsg->sonar_distance > 0){
				gpsMsg->flags.sonarOK =1;
			}
			else {
				gpsMsg->flags.sonarOK =0;
			}

bk79, 2015-07-08:
and this is the code from gps struct

typedef struct gpsDataGeneric_s {
    struct {
        unsigned gpsOk : 1;     // gps read successful
        unsigned newData : 1;   // new gps data available (lat/lon/alt)
        unsigned fix3D : 1;     // gps fix status
		unsigned sonarOK :1;
    } flags;
    int32_t latitude;
    int32_t longitude;
    uint8_t numSat;
    uint16_t altitude;
    uint16_t speed;
    uint16_t ground_course;
    uint16_t sonar_distance;
    uint16_t hdop;
} gpsDataGeneric_t;

digitalentity, 2015-07-09:
Imagine having a separate GPS unit on I2C bus (i.e. UBLOX NEO-M8N) and a separate I2C sonar unit. If we have different ways of reading sonar data from i2c-nav and other i2c-based sonars code will be a mess. Very dificult to modify and maintain. I suggest implementing a separate framework for I2C sonars.

bk79, 2015-07-09:
probably you misunderstood the scope of the code I want to implement: i2cnavboard has the capability to connect and  read internally with the actual board version and hc-sr04 cheap board, multiwii has an option to read hc-sr04 from this Navboard. what I want to do is simply to extend your driver to have the already there capability on the cleanflight code. Nav-board doesn't care about the GPS type when you flash it with config.h having ublox selected ubx will bhe protocol. Sfr02 and native i2c sonar have a completely set of different instruction on the i2c bus.  I Don't have any of those and I prefer not to buy one since they cost more than nav-board+GPS+hc-sr04 all Together and they give the same performance

digitalentity, 2015-07-09:
What you are suggesting is to make I2C-NAV sonar reading part of GPS subsystem. I understand that it is the easieast way to implement sonar support, but this will mix up GPS and sonar drivers. We will have two *different* ways of getting sonar altitude - one for i2c-nav board (occuring from gps.c and gps_i2cnav.c), another for classic hc-sr04 (occuring from sonar.c and sonar_hcsr04.c). If we will want to change anything sonar-related we'll have to remember of *both* ways.

But if we implement a separate i2c sonar driver connected to i2c-nav board (i.e. sonar_i2cnav.c) and keep all sonar-related stuff inside sonar.c, than maintaining the code would be much easier.

bk79, 2015-07-09:
Not of the GPS sub system but of GPS i2c nav. Gpsmsg  is only used by the implementation you did on the navboard  I still don't get why the typedef struct is in gps.h and not in i2cgpsnav.h

digitalentity, 2015-07-09:
In future we can have another i2c gps module supported. It will also use gpsDataGeneric_t datatype, but it will not be able to provide sonar altitude. Current code is focused on reading GPS data from some GPS module and it does not need to know about the fact that actual hardware is capable of providing sonar altitude as well.
Likewise, sonar driver will read from some sonar module (i2c or not) without knowing anything about other data besides sonar altitude. 
We have the task - "Read GPS data and sonar altitude from i2c_nav board with respect to other hardware". Now we use KISS principle - split the required task to most simple parts and keep it that way. We split this task to two: "read GPS data" and "read sonar altitude" and we solve them *separately*. This way code is more readable and easier to debug and maintain.

bk79, 2015-07-09:
OK I'll arrange the code somewhere else

bk79, 2015-07-17:
@digitalentity I've finished the driver, but I still don't have the nav board to test it, if you get both hc-sr04 and nav board please have try and let me have a feedback, it should be just a couple of commit behind yours
.https://github.com/bk79/cleanflight/tree/i2c-sonar

bk79, 2015-07-17:
@digitalentity I forgot to tell that I've set it as a feature so it should be enabled on the board and for now I've compiled only on CC3D, just add a line #ifdef SONAR_I2CNAV inside any target.h and it will be compiled

digitalentity, 2015-07-17:
@bk79 
Thanks. I should have a spare I2C-NAV board somewhere in my desk. If I find it I will definitely give it a try.

closed 2015-12-20 by digitalentity

RivetingTiger, 2017-08-15:
Hi

Just new to inav coming over from cleanflight. I am using Naze32 rev6 FULL, with GPS, OSD, SONAR.
So far GPS is working. I am stuck with the sonar. It is wired to M5 M6 as described in Naze32 setup. However can't get it to activate in iNav, using latest version, any help would be appreciated.

DzikuVx, 2017-08-15:
Sonar is not supported since naze does not have enough flash memory to fit
it .

RivetingTiger, 2017-08-15:
Thanks for your quick comment. Since no sonar, will I still get enough function out of having a gps on my quad. The Naze has built in gyro, mag and barometer? 
Sorry if it is a silly question,  but this is my first quad that I am having gps. Yet if It won't function with INAV in all functions of RTH, although hold, autonomous flight, it's not worth putting the extra weight, and I'll have to wait till next build to get an Fc that will work with GPS and sonar properly.