iNavFlight/inav #30: IMU - Migrate to a better attitude calculation algorythm

digitalentity, 2015-08-03:
NAV code is now very sensitive to direction of gravity and magnetic field vector in aircraft frame - for controlling accelerations and heading. Estimating attitude as fast and precise as possible is a must. Current complementary filter is doing well, but there are better algorythms available, i.e. Robert Mayhony's DCM filter or Sebastian Madgwick's IMU fusion.

digitalentity, 2015-08-03:
current:
   text    data     bss     dec     hex filename
 118204     224   14156  132584   205e8 ./obj/main/cleanflight_NAZE.elf

with Madgwick AHRS 
arm-none-eabi-size ./obj/main/cleanflight_NAZE.elf
   text    data     bss     dec     hex filename
 122212     220   14160  136592   21590 ./obj/main/cleanflight_NAZE.elf

Madgwick's AHRS took almost 4kb flash! I'll try see if it works better than complementary filter.

digitalentity, 2015-08-03:
Bench tests show that with Madgwick's AHRS heading is much more locked in (when using a compass). I've noticed that after a series of fast pitch-roll rotations complementary filter showed arbitrary heading and took half a second to correct using a compass. Using very same settings and same test Madgwicks AHRS was totally locked in.
But nothing comes without a price - for this I'm paying with extra 4k flash and lots of CPU cycles. When using complementary I was able to get looptime=1000, using AHRS I got only as low as looptime=1400. CPU is spending alot of time calculating AHRS.
Need to try it in-flight and see if it's worth it.

digitalentity, 2015-08-03:
Just tried to fly it - much worse than plain old complementary filter. HORIZON mode is very slow to react, copter feels mushy. That's probably due to incorrect beta used. Postponing this for now.

bk79, 2015-08-04:
Have you tried slowing loop time? Probably it could help

digitalentity, 2015-08-22:
I'm closing this as new NAV PID controllers seem to be doing their job pretty good with current IMU. Will re-open if still necessary.

closed 2015-08-22 by digitalentity