iNavFlight/inav #34: Docs - Create a PID tuning guide

digitalentity, 2015-08-20:
Seems like NAV PID controller code is stable enough and won't be changing much. NAV uses ALT, VEL, POS and POSR PIDs. We need to update docs with description what these PIDs do and how changed PID values will alter copter behaviour.

varoudis, 2015-10-05:
A basic documentation would be great in order to help you with testing out the build.
As I was not following your developments from the start and my time (probably knowledge) is limited in order to read all your commits some basic text and test patterns would be great!

In general Im interested in indoor navigation and I could probably help on that part.
Thanks for the good work.

digitalentity, 2015-10-06:
@varoudis I'm slowly working on documenting the settings. Currently I'm working on a new position estimator with the ability to do dead-reckoning - see https://github.com/digitalentity/cleanflight/issues/23#issuecomment-145319411

This is pretty much untested and need lots of improving. I'd be grateful if you can give a hand with that. Thanks!

digitalentity, 2015-10-07:
I've started a documentation project for navigation-rewrite system on GitHub. A dedicated repository is located here: https://github.com/digitalentity/nav-rewrite-docs/wiki
Don't hesitate to contribute!

varoudis, 2015-10-07:
Thanks @digitalentity that is very helpful.i also saw your rebase. Ill try to familiarise myself with the code base and start testing little hacks.

closed 2016-04-07 by digitalentity