iNavFlight/inav #37: NAV - Code review

digitalentity, 2015-08-21:
Ok, seems that I'm done with major code changes. From now on most code changes will be bugfixes. A code review is now necessary to find bugs I've missed while developing this. Also latest commits intended to solve #31 introduced some major changes and I've might have broken something in process.
@dkisselev, @csurf I would appreciate your help with this :smirk:

bk79, 2015-08-22:
Did you have any flight already?

digitalentity, 2015-08-22:
@bk79 not with the latest commits - it's too windy.

csurf, 2015-08-23:
can u please point me to a specific commit to build & test? in case you're working on stuff simultaneously, I wouldn't want to pull something that might be broken while you're still actively making changes.

digitalentity, 2015-08-23:
@csurf, I've flight tested latest changes today. You can grab and test 0b38b9e461f1663c4fd035f4358f9355f8154ffe - that's the most recent code that was flight tested, POS & POSR PIDS may require tuning.

digitalentity, 2015-08-28:
Some new changes are ready to be flight-tested. New binary **cleanflight_NAZE.navigation-rewrite.Aug28.hex** is built from 25a0e4f0e2cb47ad1729e835c476ce6633775e34 and is available in my [Dropbox](https://www.dropbox.com/sh/4r5374mos7471q2/AACppn0Mphw45p-4TJ7lZLlOa?dl=0).
Most significant changes:
* code refactoring (slight possibility I might have broken some functionality)
* re-added sonar support for low altitude ALTHOLD, needs careful testing
* added compensation for BARO drift from GPS, needs testing - enable ALTHOLD and hover for couple of minutes - shouldn't drift for more than 0.5-1 meter
* PH changes, should tolerate wind better
* RTH changes, verify that PH is working before testing this
* reworked landing code, should now descend faster.

digitalentity, 2015-09-05:
Flight tested the latest code yesterday. Looks like ALTHOLD and PH are working as they should. RTH flight test is pending - rain forced me to stop flying. New POS and POSR PIDS are working fine, some more tuning might get them to work better, but now they should be just fine for most pilots. Feel free to test c27002375a2e65b2b5df31ebd6d56e44f6b5c9fe. Also the rebase to latest official master branch is scheduled for next week.

varoudis, 2015-10-05:
is there a rebase from the master in the works?
thanks

digitalentity, 2015-10-06:
@varoudis 
There was a rebase to 52881b3dfd9bcdb2632b45fdc707b8a92aea5e40 right after @borisbstyle's LuxFloat improvments were merged.

digitalentity, 2015-11-27:
Flight testing continues in RCG thread here: http://www.rcgroups.com/forums/showthread.php?t=2495732

closed 2015-11-27 by digitalentity