iNavFlight/inav #39: NAV - Waypoint mode

digitalentity, 2015-09-06:
Hi everybody! I'm thinking about waypoint support and I wonder how the firmware should work when there is no waypoints on the list or all waypoints were successfully visited. I have two options:
1. Hold last position
2. Execure RTH

bk79, 2015-09-06:
Hold last position would be the good deal, rth could get trouble specially if in the straight line calculated from last wp to home there are some obstacles

digitalentity, 2015-09-06:
I'm thinking hold last position is a better option myself. If pilot wants RTH - just set last waypoint to home coordinates.

dkisselev, 2015-09-06:
Yeah, I'm pretty sure hold position is what most other platforms do.

In Pixhawk there's a waypoint type that triggers a landing when it arrives, so that could be a future extension.

digitalentity, 2015-09-07:
"Hold last position" it will be. I've started to implement waypoints. Added proper MSP_SET_WP message handing - EzGUI follow me feature should work now.

digitalentity, 2015-09-08:
Full waypoint navigation implemented in 16a3cb514ff86d26ced9d9482319d22adb8fb4bb - a total of 15 waypoints numbered 1 to 15 supported for compatibility with existing MSP clients where waypoint 0 is home and waypoint 16 - current hold position.

digitalentity, 2015-09-08:
It seems that MultiWii and Baseflight/Cleanflight is treating MSP_SET_WP differently. MultiWii seem to have more a advanced waypoint protocol while MSP message number is the same.

stronnag, 2015-09-08:
See https://docs.google.com/document/d/16ZfS_qwc-rJeA7N5Tx0DA6wtgxl6HdGgaz-jE3lHBWs/edit?usp=sharing for details of the mw-nav messages and data structures (as in mwp and ezgui).

digitalentity, 2015-09-09:
Ok, I've implemented a quick and dirty hack for waypoints (94b9fc271d778a9aebf56a03f17627452bf03cf6). action and flags are ignored, but lat/lon are now handled correctly.

digitalentity, 2015-12-20:
Basic waypoint support is working. Improvements should be opened as separate issues.

closed 2015-12-20 by digitalentity