iNavFlight/inav #40: ALThold mode not working on my 180 quad

skaman82, 2015-09-09:
Tested your latest naze hex form your dropbox today. Tried the ALThold mode on my 180 quad. Board: naze32 Full. When I put the quad in ALTHOLD it stays for a while but then starts slowly sinking to the ground. Could be the PIDs. The strange thing though is that the quad don't respond to throttle input while in althold mode. I could't raise the quad by increasing the throttle.

digitalentity, 2015-09-09:
Please provide some details about your setup. Blackbox log and full config
dump would be perfect.

skaman82, 2015-09-09:
I don't have a blackbox log of this. I'll try to make one next time. Never used blackbox before. So I have to figure out how this works first. My setup: Diatone ET180 Frame, Naze32 full, DYS 1306 3100kv motors, 4054 props, multistar 6A v2 ESCs with Blheli 14, OSD and FPV Gear, 950mah battery = AUW 200g. The symmetry of the frame is not an clear X - thats why I run a custom mixer.

# dump

# version
# Cleanflight/NAZE 1.10.0 Sep  9 2015 / 13:45:41 (9c10376)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer CUSTOM
mmix reset
mmix 0 1.000 -0.717 0.736 -1.000
mmix 1 1.000 -1.000 -0.736 1.000
mmix 2 1.000 0.717 0.736 1.000
mmix 3 1.000 1.000 -0.736 -1.000
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
feature LED_STRIP

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 1 115200 57600 0 115200

# led
led 0 7,12::WC:13
led 1 8,12::WC:13
led 2 0,0:::0
led 3 0,0:::0
led 4 0,0:::0
led 5 0,0:::0
led 6 0,0:::0
led 7 0,0:::0
led 8 0,0:::0
led 9 0,0:::0
led 10 0,0:::0
led 11 0,0:::0
led 12 0,0:::0
led 13 0,0:::0
led 14 0,0:::0
[... 260 lines cut ...]
set p_vel = 120
set i_vel = 45
set d_vel = 1
set yaw_p_limit = 500
set dterm_cut_hz = 0
set pterm_cut_hz = 0
set gyro_cut_hz = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 113
set rc_expo = 75
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 40
set yaw_rate = 35
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2015-09-09:
You  don't have your althold PIDs at new defaults. Specifically you have
i_alt set to 0 - that's preventing ALTHOLD from working. Please see the RCG
thread from here
http://www.rcgroups.com/forums/showpost.php?p=32643255&postcount=80 for
explanation. And execute the following from CLI:
set baro_tab_size = 7
set baro_noise_lpf = 0.350
set baro_cf_vel = 0.965
set p_alt = 30
set i_alt = 150
set d_alt = 0
set p_vel = 80
set i_vel = 50
set d_vel = 50
This will set your althold PIDs right and you should be good to do.
Good luck!

skaman82, 2015-09-09:
Thanks! Will try this and report back!

skaman82, 2015-09-10:
Tested the new PID defaults today. It works now as expected when Quad hovers normally at 50% throttle. But when I use a little heavier lipo (quad hovers then at 60%) it cant hold altitude an sinks slowly to the ground an stays 10-20 cm above the ground. What do I have to change to make this work with a  heavier lipo? By the way: like your solution! Now it works even when the quad is moving (it does not in the normal CF) and so this flightmode is finally usable as a standalone mode!

digitalentity, 2015-09-10:
@skaman82 
I'll try with heavier lipo and see if the code works for me. For now you can try increasing your i_vel, try set i_vel = 75 or 100, or even 150 and see if it gets any better. See this post http://www.rcgroups.com/forums/showpost.php?p=32643255&postcount=80 for explanation of altitude hold pids.

digitalentity, 2015-09-13:
@skaman82 
Tested today with a heavier lipo, no problems so far. My guess whould be insufficient i_vel.

bk79, 2015-10-03:
@digitalentity I finally completed my hexa 320, I had a test with brand new sparky board with latest buidl  and alt hold keeps farly well, probably there is a little too much I term since it overshoot too much, I've tested 1500mah battery and 2200mah no problem so far whatever weight I put

digitalentity, 2015-10-04:
I figure this issue can be closed as althold is verified to be working.

closed 2015-10-04 by digitalentity