iNavFlight/inav #42: AIRPLANE issues

sppnk, 2015-09-19:
I have opened issues #1323 & #1324 in main CF code, so I found that sensors stops readings when reseting the CC3D. 

I opened here because if anybody could test a CC3D board and reproduce the problems/solve them, I think here could be more people interested in fixed wings than in main CF code forums (everybody seems to play only with quadcopters :-))..

digitalentity, 2015-09-19:
@sppnk 
Thanks for opening this one. I'm currently distracted from airplanes by issues #4, #37, #41. Will return to airplane support when this will be flyable on multicopters.

hydra, 2015-09-23:
@sppnk please don't attempt to take developers away from the work they're focussed on by opening issues like this.  @digitalentity is hard at work on GPS and related code.

hydra, 2015-09-23:
@digitalentity please close this, there are issues in the main trackers that will deal with this.

sppnk, 2015-09-23:
@hydra, I don't attempt to take anybody away from nothing, but it seems the only people interested in airplanes are in digitalentity branch, so I put my issue here, just looking for more testers and feedback.

digitalentity, 2015-09-23:
@hydra I guess you are correct. These is already an issue #26 for fixed-wing support with all relevant blocking factors.
@sppnk those who interested in airplane support will likely find issue #26 and get the info there.

closed 2015-09-23 by digitalentity