iNavFlight/inav #43: GPS - expose HDOP in blackbox and MSP

alvins82, 2015-09-28:
Currently cleanflight users the number of satellites as an indicator of 'GPS health'. HDOP is a better measure of this and has been adopted by other FC platforms.

Cleanflight already has HDOP in the GPS struct (https://github.com/cleanflight/cleanflight/blob/27db9ad0fe9a4d8abc05c59db201b8944324eea4/src/main/io/gps.h#L104) but it doesn't seemed to be used or exposed. 

Ticket is two things -
- expose HDOP in MSP so it is able to be shown in the cleanflight configurator
- log HDOP in blackbox logging

digitalentity, 2015-09-28:
Current NMEA GPS driver does not seem to update HDOP. Tasks to do in this ticket:
* update GPS driver to receive HDOP and VDOP
* expose HDOP to MSP and configurator
* log HDOP to blackbox
* use HDOP/VDOP for position estimation (trust GPS less if position is inaccurate)

skaman82, 2015-11-05:
@digitalentity Could you expose also some other things to MSP so we can do that?

[image]

digitalentity, 2015-11-06:
@skaman82 How deviations are calculated?

skaman82, 2015-11-07:
@digitalentity no clue, I thought you knew... ;-)

digitalentity, 2015-11-07:
Will have to look at TC's code then :-D

marksev1, 2015-11-24:
https://github.com/multiwii/baseflight/blob/305a2fbff4cac019c382a8f86e020a58a50f8a3e/src/gps.c

Mention of DOP is in Line 44, Line 1276, comments in L 1357, L 1361, L 1487.

digitalentity, 2016-01-18:
HDOP/VDOP in the form of absolute accuracy values (EPH/EPV) are exposed to blackbox now (5a47299). 
For UBLOX and NAZA GPS these are true accuracy values in cm;
For NMEA and I2C-NAV GPS modules these are HDOP*10 values (base accuracy assumed 1m)

bk79, 2016-01-18:
Hdop and vdop are almost useless in terms of accuracy since they can be
very high in value but if you don't have enough satellites the accuracy is
less than having low hdop/vdop with lots of satellites

digitalentity, 2016-01-18:
@bk79 
DOP is not accuracy, it's **dilution of precision**, meaning low DOP = high accuracy.
Moreover I'm not using and don't intend to use HDOP/VDOP for navigation. I'm using absolute accuracy (if reported by the module). At the moment only UBLOX and NAZA are reporting true accuracy values.

closed 2016-03-18 by digitalentity