iNavFlight/inav #45: mwosd compatibility

skaman82, 2015-10-10:
I tested your 0710 build today. It would be extremely helpful to show current flightmode in mwosd like it already works in cleanflight master.

• when I activate ALTHOLD the OSD does not sho baro sensor active
• in POSHOLD we should see the "Position hold" message like in master
• in RTH we should see the "return to home" message like in master

Maybe you could integrate this functionality or simulate the master behavior.

digitalentity, 2015-10-10:
Confirmed, will make things compatible to master soon.

digitalentity, 2015-10-11:
b67a513 possibly fixed the issue. Upcoming 1011 build will include this fix.

digitalentity, 2015-10-17:
@skaman82 can you verify that newer builds are compatible?

skaman82, 2015-10-17:
Try to test this today and will get back to you.

skaman82, 2015-10-17:
Tested this with the 1012 build just jet. OSD now shows BAR active when in ALT mode and the RTH and POSHOLD massages are displayed as well. Wonderful. But there are some tiny pieces missing to make it perfect. 

• ALTHOLD: BAR icon shows only if armed - this works in master without arming
• POSHOLD: MAG icon is not active
• RTH: MAG and BAR icons are not active

Tested all the things I mentioned in #46 – all wrks very great now. No altitude drops anymore, full throttle control in POSHOLD. Really impressed with your work! Thank you.

digitalentity, 2015-10-17:
Good! Thanks for testing this for me. Did POSHOLD work well for you?

New code treats modes differently. For one, modes are enabled only when armed and possible to activate (i.e. PH won't activate if no GPS fix). This is intended behaviour. 

MAG will never be active when in PH/RTH modes, my code has its own heading lock controller. This, however, might change in future.

RTH is a stand alone mode, it does not activate a separate BARO mode, altitude is controlled internally by navigation code without exposing the fact to the outside.

To conclude: mwosd now shows exactly what new code is doing. Pilot just have to remember that POSHOLD and RTH always control heading, and RTH always control altitude if there is an altitude sensor (baro).

I think we can consider the issue resolved?

skaman82, 2015-10-17:
OK I guess we can close this. POSHOLD worked very well! Quad was not moving around a bit and landing in RTH was always spot on within 1m from take off spot. Almost no wind today.

closed 2015-10-17 by skaman82

digitalentity, 2015-10-17:
Thanks! New pos estimator seems to be doing a great job. 
Big change to the code is coming soon. It weather permits I'll flight-test it tomorrow before uploading a new build.

skaman82, 2015-10-17:
This is the test vehicle BTW :) 
[image]

digitalentity, 2015-10-17:
A great looking quad! Let's make it fly better than DJI Phantom!

skaman82, 2015-10-17:
sounds good to me