iNavFlight/inav #46: Feedback of 0710 build

skaman82, 2015-10-10:
I tested your 0710 build today with my 350 class quad. naze32 full (mag removed), neo-8m GPS and external compass. Average satellite count: 12. AUW 1.2 kg. Hovering point: 63% throttle.

• ALTOHOLD works like a charm. I love it!

• POSHOLD works for me ok…ish. The quad starts to fly in circles after a short while. The radius increases over time. Calibrated my compass three times but the behavior didn't change. It was little windy - 12 km/h.

• I little confused about RTH mode: Does it should keep the height by itself? When I activate RTH the quad starts immediately sinking until is hits the ground. It should come back to me and then attempt to land, right? nav_rth_alt_mode is set to 2, nav_rth_altitude is set to default 1000 (is this in cm or mm?). When I activate RTH and ALHOLD the quad is flying back in my direction but drifts to the right. It worked pretty precise in CF 1.9. I maybe have to check my compass, but I didn't explain the sinking to the ground in RTH. Tested all modes in 5-10m height and 20-40 m radius.

# version
# Cleanflight/NAZE 1.11.0 Oct  7 2015 / 07:05:31 (ba11933)
# pflags
# Persistent config flags: 0x00000001
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature GPS
feature FAILSAFE
feature TELEMETRY
feature LED_STRIP

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 115200 0 115200

# led
led 0 3,2::C:6
led 1 2,1::WC:6
led 2 1,0::C:6
led 3 3,5::C:2
led 4 2,6::WC:2
led 5 1,7::C:2
led 6 5,5::WA:0
led 7 6,5::WA:0
led 8 8,5::C:2
led 9 9,6::WC:2
led 10 10,7::C:2
led 11 8,2::C:6
led 12 9,1::WC:6
led 13 10,0::C:6
led 14 0,0:::0
led 15 0,0:::0
led 16 0,0:::0
led 17 0,0:::0
led 18 0,0:::0
led 19 0,0:::0
led 20 0,0:::0
led 21 0,0:::0
led 22 0,0:::0
led 23 0,0:::0
led 24 0,0:::0
led 25 0,0:::0
led 26 0,0:::0
led 27 0,0:::0
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = 0
set rc_smoothing = 1
set input_filtering_mode = 0
set min_throttle = 1050
set max_throttle = 1900
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 0
set auto_disarm_delay = 10
set small_angle = 25
set reboot_character = 82
set gps_provider = 1
set gps_sbas_mode = 1
set gps_auto_config = 1
set gps_auto_baud = 0
set inav_dead_reckoning = 0
set inav_gps_delay = 200
set inav_w_z_baro_p =  1.000
set inav_w_z_baro_v =  0.500
set inav_w_z_sonar_p =  2.000
set inav_w_z_sonar_v =  1.000
set inav_w_z_gps_p =  0.300
set inav_w_z_gps_v =  0.300
set inav_w_xy_gps_p =  1.000
set inav_w_xy_gps_v =  2.000
set inav_w_xy_dr_p =  0.050
set inav_w_xy_dr_v =  0.050
set inav_w_z_res_v =  0.500
set inav_w_xy_res_v =  0.500
set inav_max_eph_epv =  1000.000
set inav_sonar_epv =  20.000
set inav_baro_epv =  100.000
set nav_use_midrc_for_althold = 1
set nav_throttle_tilt_comp = 1
set nav_lock_until_takeoff = 0
set nav_user_control_mode = 0
set nav_pterm_cut_hz = 20
set nav_dterm_cut_hz = 15
set nav_wp_radius = 200
set nav_max_speed = 500
set nav_manual_speed = 500
set nav_manual_climb_rate = 200
set pos_hold_deadband = 20
set alt_hold_deadband = 40
set nav_min_rth_distance = 500
set nav_rth_alt_mode = 2
set nav_rth_altitude = 1000
set serialrx_provider = 1
set spektrum_sat_bind = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 35
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set current_meter_type = 1
set align_gyro = 0
set align_acc = 0
set align_mag = 2
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 350
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set pid_at_min_throttle = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = 1
set servo_lowpass_freq = 400
set servo_lowpass_enable = 0
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1400
set failsafe_kill_switch = 0
set failsafe_throttle_low_delay = 100
set rx_min_usec = 985
set rx_max_usec = 2115
set acc_hardware = 0
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = 1
set magzero_x = -18
set magzero_y = -281
set magzero_z = 22

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

# aux
aux 0 1 0 900 2100
aux 1 23 0 1500 2100
aux 2 24 1 1350 1475
aux 3 25 1 1850 2100
aux 4 9 1 1850 2100
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 90
set gps_posr_i = 15
set gps_posr_d = 1
set gps_nav_p = 0
set gps_nav_i = 0
set gps_nav_d = 0
set deadband = 0
set yaw_deadband = 5
set throttle_correction_value = 0
set throttle_correction_angle = 800
set default_rate_profile = 0
set gimbal_mode = 0
set acc_lpf_factor = 4
set acc_unarmedcal = 1
set acc_trim_pitch = -8
set acc_trim_roll = -6
set baro_tab_size = 7
set baro_noise_lpf =  0.350
set mag_declination = 200
set pid_controller = 3
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  1.600
set i_pitchf =  0.300
set d_pitchf =  0.030
set p_rollf =  1.600
set i_rollf =  0.300
set d_rollf =  0.030
set p_yawf =  5.500
set i_yawf =  0.150
set d_yawf =  0.050
set level_horizon =  3.000
set level_angle =  5.000
set sensitivity_horizon = 75
set p_alt = 30
set i_alt = 150
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 80
set i_vel = 50
set d_vel = 50
set yaw_p_limit = 500
set dterm_cut_hz = 0
set pterm_cut_hz = 0
set gyro_cut_hz = 0
set pid5_oldyw = 0

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 50
set rc_yaw_expo = 0
set thr_mid = 55
set thr_expo = 10
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 25
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2015-10-10:
Hi! Thanks for the feedback. Going in circles can be caused either by bad
compass readings  or overshooting due to high pids.

Can you try zeroing posr_d and reducing posr_p to 50 and see if it's
working better for you? Also, where are you located? You mag declination
might be off and cause heading to be off as well.

-- 
Sent from Gmail Mobile

skaman82, 2015-10-10:
Thanks for the advice. I will test this tomorrow and report back. I live in Hamburg, Germany so the mag declination should be fine.

digitalentity, 2015-10-10:
Yes, mag declination seems right. Lowering PosR PIDS should help. If it doesn't, you'll probably have too much magnetic interference or bad compass calibration. 

There is an issue on improving mag calibration https://github.com/cleanflight/cleanflight/issues/1372 , should make things easier.

skaman82, 2015-10-11:
Tested your 1010 build today. Configured everything from scratch. ALTHOLD works fine as expected.

POSHOLD had some troubles. Quad holdes position quite well now - I'm guessing I messed some variables up last time. It was very windy today so position hold wann't spot on but much better than the last time (no flying in circle anymore). 

But I had also some problems. After activating just POSHOLD the quad dropped down to the ground violently two times. I don't know if it has anything to do with my quad hovering at 63%. But most of the times it kept hovering at the previous altitude. I noticed that I have absolutely no throttle control in POSHOLD - just pitch, roll and yaw. I wasn't able to take off in POSHOLD mode. I think this ist not so good. It would be better to have full throttle control. But not like in ALTOHOLD mode where throttle input is not so direct. So we can move freely up and down but the quad stays in the same position.This would be great for filming. I think this is how DJI is doing it but I'm not sure. If we want the quad to maintain altitude we could simply add the ALTHOLD mode.

When I activated POSHOLD + ALTHOLD I had control over the altitude. The trottle input behaves than like it normally does in ALTHOLD.

I also noticed that on stick input the quad moves much slowly than in normal ANGLE. 

Then I moved on to testing the RTH. This time it worked flawlessly. It was a little scary when the quad is returning. It has much more speed than it does in master. Is there a variable that slows this down a little. It is very cool that the quad performs a 180° turn twice. It turns around (nose to home position), comes back to me, turns again 180° and attempts to land. But the turning could be potentially dangerous for rookies. When something goes bad while RTH you have to rethink controls.

Landing is already quite good. Love this feature! The quad hit the ground (softly) twice before it finally landed. The landing spot wasn't exact the takeoff sport but within 4m radius. That might be because of the wind today. Have to test this without wind.

digitalentity, 2015-10-12:
I've had reports of altitude drop when enabling only POSHOLD and opened the separate issue (#44), will look into it soon.

I've also noticed that stick input in POSHOLD/GPS_ATTI is treated differently than in normal ANGLE. That also need fixing - opened an issue #47 to keep track of this issue.

Thanks for testing RTH, that's the very first RTH test with new iNav estimator. Good to know it worked for you. You can tweak the speed by changing nav_max_speed from CLI (in cm/s). Default 500 cm/s is probably too high, I'll lower it in the next build.

I'm already thinking of re-adding an option to return home tail-first (will be easier for novice users)./

closed 2015-10-17 by digitalentity