iNavFlight/inav #48: improving ublox  gps accuracy

bk79, 2015-10-17:
@digitalentity I've noticed that ublox from main streamline uses 5Hz update rate, even if you set manually the gps settings to 10Hz whenever you start back the FC it will automatically reset it to 5Hz,
I've noticed that the incriminated lines are inside ..\io\gps.c, changing the line 150 with
this "0xB5, 0x62, 0x06, 0x08, 0x06, 0x00, 0x64, 0x00, 0x01, 0x00, 0x01, 0x00, 0x7A, 0x12," will make the trick

digitalentity, 2016-01-23:
I'm closing this, a very popular Neo-6M does not support 10Hz update rate. With version detection working Neo7 and above will be set to 10Hz rate already.

closed 2016-01-23 by digitalentity

bk79, 2016-01-23:
Actually I've tried it several times and 10hz with neo6m works perfectly
giving more accuracy

digitalentity, 2016-01-23:
It works with my 6M too, but I have reports of even M7Ns having issues with 10Hz updates. I'd rather stick to the datasheet on this one :smile: