iNavFlight/inav #58: Filsafe event

brat002, 2015-11-16:
Could you add position hold as configurable action on failsafe event?

digitalentity, 2015-11-16:
Current upcoming task is RTH on failsafe event. Why would you want poshold on lost radio?

brat002, 2015-11-16:
In LOS or near by LOS conditions you can restore control just by antenna alignment. No need RTH fo all. Automatic RTH may be dangerous on long ranges with many building and trees.
As proposal - fire PH for failsafe on "configurable timeout seconds" and then RTH. It will be safe for most.

Maybe auto landing in RTH mode will be good solution also.

digitalentity, 2015-11-16:
Interesting idea. I'll consider entering poshold for some time (configurable) and then execute normal RTH sequence.

MauroMombelli, 2015-11-16:
Instead of timing, if battery status can be estimated I would use that.
Also a "soft landing in place" option may be useful.

digitalentity, 2015-11-16:
Thanks for the idea. First of all we need to get RTH working reliably, then go for advanced failsafe mechanisms.

digitalentity, 2015-12-20:
RTH on RX signall loss is working now (tested)

commit 5d5fd81b (2015-09-11): Fix bug where PPM rx stops working on sparky or CC3D when using motor
PWM rate > 500 (brushed motor mode).

Fix is the same as for OneShot since both brushed motors and oneshot use
an 8mhz timer.

Fixes #58
closed 2015-12-20 by digitalentity