iNavFlight/inav #59: NAV - Add V-tail Airplane mixer support

marksev1, 2015-11-24:
Add support for V-tail airplane. Thus covering the entire range of airplane configurations that are used the most.

Ralf-W, 2016-03-14:
I think it could work using a custom mixer. Starting from the examples given in CF doc "Mixer.md" I just tried the following:

 mixer CUSTOMAIRPLANE
 mmix reset
 mmix 0 1.0 0.0 0.0 0.0 # Engine

 smix reset
 # Rule Servo Source Rate Speed Min Max Box
 smix 0 3 0 100 0 0 100 0 # Roll / Aileron
 smix 1 4 0 100 0 0 100 0 # Roll / Aileron
 smix 2 2 1 100 0 0 100 0 # Pitch / Elevator
 smix 3 5 1 100 0 0 100 0 # Pitch / Elevator
 smix 4 5 2 -100 0 0 100 0 # Yaw / Rudder
 smix 5 2 2 100 0 0 100 0 # Yaw / Rudder

In CF "Motors"-tab it does exactly what I expect it to do for a plane with ailerons and a V-tail. However I did not try it on a real model plane.

Ralf-W, 2016-03-17:
Correction! Stupid me! I was misled by the examples given in the CF docs "mixer.md". What I wrote above is OK for all the servos. But "Throttle" should also be defined using "smix": as a servo signal led to the first output pin, where the ESC will be connected.

I think it should work using this a custom mixer:

mixer CUSTOMAIRPLANE
 mmix reset
 
 smix reset
 # Rule Servo Source Rate Speed Min Max Box
 smix 0 3 0 100 0 0 100 0 # Roll / Aileron
 smix 1 4 0 100 0 0 100 0 # Roll / Aileron
 smix 2 2 1 100 0 0 100 0 # Pitch / Elevator
 smix 3 5 1 100 0 0 100 0 # Pitch / Elevator
 smix 4 5 2 -100 0 0 100 0 # Yaw / Rudder
 smix 5 2 2 100 0 0 100 0 # Yaw / Rudder
 smix 6 0 3 100 0 0 100 0 # Throttle 

In CF "Motors"-tab it does exactly what I expect it to do for a plane with ailerons and a V-tail. However I still did not try it on a real model plane.

digitalentity, 2016-03-18:
@Ralf-W 
Please see my comment here https://github.com/iNavFlight/inav/issues/80#issuecomment-193619370
I'm actually thinking about **removing** exotic mixes from pre-defined and only allow them to be defined as "custom". I think non-common mixers setups should go to the wiki.

closed 2016-04-01 by marksev1

oleost, 2016-04-01:
@marksev1 @Ralf-W 

Can any of you confirm that the custom mix works? If so I, or one of you can add it to the iNav wiki as a new page "Custom Mixes for fixed wing and multirotors"

Ralf-W, 2016-04-01:
@oleost I still have not flown a model with it but my naze32 does exactly what I expect it to do on the bench. 

However, please have a look at the comment of  @digitalentity: He is not amused about using exotic mixes because they block too much memory. As I understand CUSTOMAIRPLANE will not survive in iNAV. Thus I will try mixer CUSTOM and report soon.

digitalentity, 2016-04-01:
@Ralf-W CUSTOM, CUSTOM_TRI and CUSTOM_AIRPLANE mixers will survive. It's non-custom exotic mixers like BICOPTER, ATAIL, VTAIL etc. that will go.