iNavFlight/inav #62: Geofence option

digitalentity, 2015-12-20:
RTH if > specified distance from home

DzikuVx, 2017-01-04:
Also, in case of "fly away" add a condition: if geofence triggered RTH but UAV is still going away, trigger emergency landing

JWC1304, 2018-07-08:
turn to home after XXXX meters is not geofence!
I have found some C# code for geofence. Still need a gui or export or import functions from google map as user interface.

public bool Contains(GeoLocation location)
{
    if (!Bounds.Contains(location))
        return false;

    var lastPoint = _vertices[_vertices.Length - 1];
    var isInside = false;
    var x = location.Longitude;
    foreach (var point in _vertices)
    {
        var x1 = lastPoint.Longitude;
        var x2 = point.Longitude;
        var dx = x2 - x1;

        if (Math.Abs(dx) > 180.0)
        {
            // we have, most likely, just jumped the dateline (could do further validation to this effect if needed).  normalise the numbers.
            if (x > 0)
            {
                while (x1 < 0)
                    x1 += 360;
                while (x2 < 0)
                    x2 += 360;
            }
            else
            {
                while (x1 > 0)
                    x1 -= 360;
                while (x2 > 0)
                    x2 -= 360;
            }
            dx = x2 - x1;
        }

        if ((x1 <= x && x2 > x) || (x1 >= x && x2 < x))
        {
            var grad = (point.Latitude - lastPoint.Latitude) / dx;
            var intersectAtLat = lastPoint.Latitude + ((x - x1) * grad);

            if (intersectAtLat > location.Latitude)
                isInside = !isInside;
        }
        lastPoint = point;
    }

    return isInside;
}

teckel12, 2018-11-17:
A "hard" geofence distance (and height) would be nice to keep legal. As long as it's turned off by default. I live in a very flat area, so I would probably turn on the height fence to say 450 feet. We already know distance to home, so it's just a matter of how to deal when exceeding the max altitude or distance.

hugsgui, 2018-11-17:
I saw in the Pawel spychalski's YouTube channel that he is working on a kind of bracking, for the next firmware, could be the option that the distance max would be triggering, cause for the max height, althold is already able to maintain the quad at a specific altitude.

teckel12, 2018-11-17:
Maybe "hard" and "soft" geofence options? "hard" would trigger RTH, while "soft" would trigger position hold and throttle max speed until your back inside the geofence? Maybe even a combination, for example, soft geofence at 0.8km and hard at 1.0km. Obviously, there would be OSD to notify of both.

hugsgui, 2018-11-17:
I'd really like something like that, so I could work with my own Inav setup, would be amazing...

hugsgui, 2019-02-01:
Hi! Now this request seem to be open about 3 years ago, any news about a geofence option in INAV? Does any of the Devs could answer about it? Would be nice to know if we are waiting for something or if I should build an ardupilot setup even if I rather use INAV.

ianrmurphy, 2019-08-18:
This quad claims Geofencing using the YupiF7 FC: https://www.studiosport.fr/racer-corsair-s1-s3-studiosport-a17684.html  also see Fincky video here:  https://www.youtube.com/watch?v=8OJxaEc2Sbg

YupiF7:  https://www.yupifc.com/

Anyone know if it's using an iNav fork and if so, where to find the source?

ianrmurphy, 2019-08-27:
OK understood - can't seem to find it though... Is the source published?

mStrangers, 2019-08-27:
betaflight fork not inav one

mStrangers, 2019-08-28:
as i know ,not published and it is not planed .

ianrmurphy, 2019-08-29:
Is that OK ref. GPL? If there was published source I would try and port to iNav...

etracer65, 2019-09-18:
> Is that OK ref. GPL? If there was published source I would try and port to iNav...

No, it's not ok. It's a violation of GPL.

mStrangers, 2019-09-20:
in fact i don't think this violating the GPL ,
But i'm not a GPL specialist and this isn't my code.

etracer65, 2019-09-20:
> in fact i don't think this violating the GPL , But i'm not a GPL specialist and this isn't my code.

A derivative work based on a GPL licensed project (in this case Betaflight) that's being distributed without publishing the source code... That's the very definition of a GPL license violation.

ianrmurphy, 2019-09-20:
Possibly what's happening here is that YupiFC distribute their binary (in the flight controller) and then offer the source only to those who a) have the binary and b) request it.
I'm no GPL specialist either, but that may actually satisfy the GPL because I don't think the GPL requires broad publication of the 'derivative work' source just access to it to those entitled (i.e. those who have the binary).

etracer65, 2019-09-20:
Nope, doesn't work that way. You can't withhold the source or otherwise charge for it (by only providing it to customers) and it must be freely available to anyone who wants it. Applying additional license terms to the GPL licensed software (such as requiring purchase or other terms of use) is expressly prohibited. By not providing the source of the derivative work they are not authorized to use the original licensed source. Period. The entire point of the GPL license is that it requires the propagation of the open source aspect. In a nutshell, by taking advantage of others' work for your benefit you're agreeing to contribute your improvements to further the project. You probably should read the GPL license terms for a better understanding.

https://www.gnu.org/licenses/gpl-3.0.en.html

rhass, 2020-07-11:
This feature should also include support for exclusion zones within a geofenced area. For example, I fly at a field which has communication towers on the perimeter. If the tower interferes and triggers a failsafe return-to-home, I wouldn't want the it to fly the vehicle directly into or through the RF beams of the tower.

Insorior, 2020-11-03:
Hi !
Has this evolved at all ?
I mean right now flying a drone professionally in France is forbidden if you can’t limit it’s operation verticaly AND horizontally. BF doesn’t do it, and arducopter, meh ! I’d better find a way to implement it on existing builds rather than having to swap hardware and I’m pretty sure that everyone does. It’s critical and should be implemented IMHO @DzikuVx

Tdogb, 2023-03-08:
This can be done using the inav programming tab now. Using the "home distance" and override rc channel to enable rth. You can also add a sticky in there to only enable rth between 50m and 49m from home.

rhass, 2023-03-08:
@Tdogb You have a good suggestion.  I think the key feature of this request is the user defined polygon. There may be very specific areas boarding a flight area which which are closer than other parts of the flight area, and should trigger an RTH. Eg: Trees, buildings, terrain, etc..

JWC1304, 2023-05-02:
any developments?!

closed 2023-10-27 by DzikuVx