iNavFlight/inav #70: UBLOX GPS won't be detected

HO4X, 2016-01-05:
Hey Guys, 

this day i am updated my NAZE 32 FULL to INAV 1.0 RC3 and had to notice that my gps-module ( drotek M8N @ URAT2 - 57600BAUD - UBLOX Protocol ) wont show up, also i could not arm my quad. I have done a 6-Point calibration and off course, a mag calibration. Iam using the newest cleanflight-configurator (1.1.0). When i flashed the "stock" cleanflight-firmware, there were no problems or something else. Before this update i used a older build of inav and everything worked pretty well. By the way, great project !!

stronnag, 2016-01-05:
The issue I'm seeing is not that the GPS is not detected, it's that it takes an inordinate amount of time from cold boot (a soft boot will usually report detection very quickly). Otherwise patience is needed, as the following log shows (given the time between power on and initial connection, the delay until detection is probably nearly 60 seconds).
````
$ mwp -s /dev/ttyUSB0
2016-01-05T17:12:17+0000 mwp startup
2016-01-05T17:12:17+0000 libchamplain 0.12.12
2016-01-05T17:12:17+0000 Starting proxy thread
2016-01-05T17:12:35+0000 Connected /dev/ttyUSB0
2016-01-05T17:12:35+0000 set mrtype=3
2016-01-05T17:12:35+0000 cap =3c
2016-01-05T17:12:36+0000 Masks arm 1 angle 2 horz 4 ph 100 rth 200 wp 400
2016-01-05T17:12:36+0000 Sensors: Acc Baro Mag
2016-01-05T17:12:36+0000 message => No GPS detected
2016-01-05T17:12:36+0000 alert file:///home/jrh/.local/share/mwp/beep-sound.ogg
2016-01-05T17:12:36+0000 no gps, sensor = 0x7
2016-01-05T17:12:36+0000 Timer cycle for 4 (0ms) items, 25 => 54 bytes
2016-01-05T17:12:36+0000 Start poller
2016-01-05T17:12:56+0000 GPS appears
````

stronnag, 2016-01-05:
And if I replace the new `ubloxInit6[]` with the old pedestrian `ubloxInit[]` (227 v. 179 bytes), it makes NO difference, so it's not as it we're overwhelming the device the init data (i.e. there is still a significant delay in detection, at least 20 seconds in the common case).

digitalentity, 2016-01-05:
I think I know the cause - init data is sent to GPS module 1 byte per call
to gpsThread.

Wasn't a problem until new scheduler slowed gpsThread down to 10Hz making
GPS init data being sent at only 10 bytes per second!

We need to send init data in larger chunks to make init happen faster.

GPS autobaud doesn't work for the same reason - timeout triggers before GPS
init data is fully sent.

stronnag, 2016-01-05:
10 secs a sec for 227 bytes would tie in exactly with the minimum detection delays I'm seeing

sambas, 2016-01-05:
Yeah, I can confirm this. Loaded RC3 to my cc3d and scoped gps configuration output. Seeing one byte at time 100ms interval. Didn't notice this because I don't have new scheduler running yet.

stronnag, 2016-01-05:
Confirmed, an ugly 0..9 loop around the init and sbas init parts fixes it.

digitalentity, 2016-01-05:
Good, now we know what's causing the regression. We probably need a cleaner
fix than a dumb 0-9 loop. Can't do any coding on my cellphone, sorry :)

stronnag, 2016-01-06:
for the record, here's the dump patch, until be have a unified scheduler / new GPS infrastructure build.
[gps_patch.diff.txt](https://github.com/iNavFlight/inav/files/80328/gps_patch.diff.txt)

digitalentity, 2016-01-22:
@HO4X 
Can you please verify if your GPS is working fine with 1.0-RC5?

HO4X, 2016-01-22:
Yes ! I will do it later that day.

closed 2016-01-24 by HO4X

digitalentity, 2016-01-24:
@HO4X thanks!