iNavFlight/inav #73: IMU - Accel and Gyro temperature compensation

digitalentity, 2016-01-18:
I've been testing CC3D outside today and noticed that estimated attitude drifted a lot after I move the board from the room (+20 deg) to the outside (-25 deg). And by a lot I mean almost 70 degrees.

**imu_dcm_ki** was doing great job canceling gyro temperature bias, but the problem was accelerometer itself. It was also offset by low temperature and that was confusing the IMU.

We need a way to compensate for this temperature drift.

MauroMombelli, 2016-01-18:
this is a know issue of the sensors, many say to be already taking care of
it. (temp. compensated)
Fixing that is quite hard has you have to create the specific drift curve
based on temperature, and from here fix the reading; this require some sort
of temperature controlled chamber witch provide at least 3 different
temperature point to recreate the curve; also you'll need some sort of
feedback from the temperature from the board, better if come directly from
the sensor (another great issue is while the sensor is changing
temperature, because it does not happen instantaneously).

hope someone can tell me wrong

digitalentity, 2016-01-19:
Today's test:
From +20 to -25 deg.
Gyro offset drift: (-2; 4; -2) degrees per second
Acc offset drift: (0.05; 0.03; 0.1) G

Enough to cause attitude drift for 1-2 degrees, makes acro flying impossible as gyro bias is causing uncontrolled rotation.

brat002, 2016-01-21:
You can get Taulabs temperature compensation code for these needs.

closed 2023-10-27 by DzikuVx

cptX, 2024-11-03:
I have a 3" multicopter where the heat from the VTX causes drift of the MPU6000 up to 10 degrees! Turning the VTX Off the accelerometer comes back to 0 so I see a clear correlation between temperature and drifting... I believe we should implement thermal compensation.