iNavFlight/inav #74: Failsafe - Voltage-activated failsafe

digitalentity, 2016-01-22:
Opened to track https://github.com/cleanflight/cleanflight/issues/1711

DzikuVx, 2016-04-13:
In 6094feb19455da372b3e0de3025c9ba529cc5ce9 I've implemented, in a little hacky way, battery injected as additional AUX channel (AUX9) as @ledvinap suggested in cleanflight#1711
ATM it's working only on NAZE and SPRF3 in PPM mode since those only targets I'm able to test. 
This allows to activate any failsafe (RTH, Failsafe switch) as a regular mode select in Configurator. 
Perhaps some additional prevention should be applied? @digitalentity what do you think?

Not live tested yet, maybe this weekend.

digitalentity, 2016-04-17:
Hi, this is interesting. However there must be a certain level of hysteresis to prevent failsafe switching on and off when your AUX9 is at the edge of the configured action.

DzikuVx, 2016-04-17:
This is still experimental. I was thinking about smth. more drastic than hysteresis. Vbat is already filtered, but there is issue of battery "regaining" voltage when load has lowered and thus exiting failsafe mode. 
So, my idea I will test next week, is to allow virtual AUX to go only down, not up

digitalentity, 2016-04-17:
Yes. I'm thinking about tracking a minimum voltage (minVbat) and apply some sort of a filter. I.e. update the minVbat only when (vbat < minVbat) for at least 2-3 seconds. This will assure short voltage drops (i.e. hard banking or counteracting wind gusts) won't trigger anything.

bk79, 2016-11-28:
@digitalentity any hopes to have this in the next release?

digitalentity, 2016-11-28:
@bk79 not sure. We need to make this flexible enough to be useful for vbat or consumed mAh.

digitalentity, 2016-11-28:
@DzikuVx what's your oppinion. IIRC you were hacking something together?

DzikuVx, 2016-11-28:
There is experimental https://github.com/iNavFlight/inav/tree/battery-voltage-as-virtual-aux that exposes Vbat as AUX channel
But this is not a good approach the way I see it now. Proper solution would have to be designed.

bk79, 2016-11-28:
I think that a simple trigger o vbat level would suit, current is not the correct approach since capacity changes from battery to battery and also during the time, but the voltage level is more or less giving the correct indication of charge % despite the age of the battery. I would be more than happy to try if you make cli parameter to activate a safe landing on a vbat low threshold

DzikuVx, 2016-11-28:
Simply exposing Vbat as AUX is absolutely not user friendly. While I'm fine with trigger based on Vbat, special GUI and dedicated configuration is required

bk79, 2016-11-28:
@DzikuVx yes and it would be safe only to activate emergency  landing RTH needs probably a little more thinking since it would involve an evaluation of the straight distance from the actual location to where the home is

oleost, 2016-11-28:
Could also be a simple two stage failsafe at voltage.

1. At configured voltage start RTH.
2. At another configured voltage start emergency land.

bk79, 2016-11-28:
@oleost something like warning level RTH if during RTH voltage drops too much emergency level will trigger landing

bk79, 2016-12-15:
@digitalentity any news on this one?

danarrib, 2017-03-02:
Ok, I had this situation yesterday. I totally forgot about battery and had to "emergency land" my Flying wing without motors. OSD was blinking battery for 2 minutes before I lose motor (ESC automatically shut down motor when voltage drops bellow 3v per cell).

It'll be awesome if INAV just trigger the Failsafe procedure after - let's say - 1 minute of voltage below a specified value (but obviously giving pilot some way to override this behavior if needed).

oleost, 2017-03-19:
@digitalentity cleanflight#1711 is closed. Tried to get.a comment from Hydra but he did not respond.

digitalentity, 2017-03-25:
Doesn't mean that we don't need the low-battery failsafe :)

oleost, 2017-03-28:
#1475

teckel12, 2017-08-15:
I really like this idea!  Have we considered having it based on the mAh used instead of voltage?  Basically using `calculateBatteryCapacityRemainingPercentage()` along with the battery_capacity variable.  Obviously, a current sensor would be required, but using this method is very linear and doesn't sag like voltage.

I use a custom build of iNAV where I send the fuel percentage (based on mAh) via telemetry.  I then created a Lua script on my transmitter that looks at the telemetry distance and percent battery left (based on mAh) and makes a decision as to when RTH should happen.  I have a it setup to warn me first, and if I don't return it returns on it's own.

In any case, virtual AUX channels in general sounds like an excellent idea.  And one which you could then set a flight mode mode or RTH if you wish right from the configurator that anyone could setup without any programming.  Also, while I do the pre-math on the RTH climb/travel/descend speed to calculate how long it will take to get back to home and what my average mAh per second draw is, if this was done on the FC it could do this calculation based upon current nav speed variables set and be more accurate.

Maybe I'll refactor "Vbat as AUX channel" to "Battery capacity remaining as AUX channel" on my own and see how it works.

snakeeater12, 2019-05-14:
Such a great idea Teckel12!  
Did you manage to get it working so when the battery gets low the drone RTH automatically as a selectable mode on I nav?