iNavFlight/inav #76: [Feature]add MavLink Code Telemetry to use Tower App

Linjieqiang, 2016-01-28:
First,Thanks your hard wroking.The Code was very nice.But I suggest quickly add the Mavlink Support so I can use Tower (https://github.com/DroidPlanner/Tower) to control the copter like APM!

digitalentity, 2016-01-29:
Hi, thanks for your interest in iNav.
Unfortunately telemetry in iNav doesn't provide means to control the copter. It's one-way communication from FC to OSD/Ground.

Linjieqiang, 2016-01-29:
But why not remove CLI and replace by Mavlink control?

digitalentity, 2016-01-30:
Removing CLI is a good idea, but replacing it by MavLink is definitely NOT a good idea - we already have MSP, why reinvent the wheel for pushing settings to the FC?

digitalentity, 2016-01-30:
A Mavlink-MSP proxy might be a solution in your case.

Linjieqiang, 2016-01-30:
Oh ,I know.Thanks

closed 2016-01-30 by Linjieqiang