iNavFlight/inav #80: Sync/catch-up with Cleanflight

digitalentity, 2016-02-08:


digitalentity, 2016-03-06:
I was unavailable to be dev for a while and I need to catch up.
What are features that present in latest Cleanflight but not in iNav?

stronnag, 2016-03-06:
From betaflight, please consider the new beeper command, just to shut up the damn thing up on USB, as I spend quite a lot of time re-flashing / reconfiguring.

digitalentity, 2016-03-06:
Thanks, I will.

dkisselev, 2016-03-07:
Is there a plan for iNav in relation to CF?

After all is said and done, is it in a state that can be merged back and become part of CF, or does it have too many specific optimizations/changes that would probably be better suited as a parallel project that will just get synced up every few releases?

digitalentity, 2016-03-07:
I am considering keeping iNav a separate project.

Reason for that is that Cleanflight targets racing machines and attempts to squese racing-related features and optimisations that compete with navigation for CPU time and FLASH memory space. It might not be possible to fit all of them into a older board like NAZE and I am almost certain that Hydra's call would be for racing-specific feature, not for navigation.

At this point I'm not going to completely separate this project for CF, but I'm not sure that all of the new code from CF will make it to iNav.

skaman82, 2016-03-07:
Konstantin, I fully agree on that. I am really tired of the direction CF is going. It almost seems that the devs think that racing is the only reason to fly a quad... and that there is only one real flight mode: ACRO. I personally would like to use it also for areal filming and navigation. The is also now a new branch called Raceflight. So all branches CF, Betaflight and Raceflight are all about racing. Where the sense in that?

I really hope that iNAV delivers that precision that a DJI Phantom is offering and I understand and support your decision fully. Have iNav flashed to most of my quads and the are flying much better in leveled flight modes than with all other branches. Especially with 1.12 somehow I really have to fight drifting issues.

digitalentity, 2016-03-07:
@skaman82
1.12 and betaflight sacrificed full accelerometer sampling rate in favor of top gyro performance and how suffer consiquences of aliasing the accel data. iNav has already been down this road a few months ago soon after new scheduler was introduced.

I let racing pros fly two identical quads, tuned to the edge of acro performance, one flashed with Betaflight (1kHz looptime) and the other with the custom-built code with filtering sitmilar to iNav's running at 500Hz looptime. They said that it was flying almost at good as Betaflight. I wonder what would be the result if I did a blind test - would they be able to distinguish between the two?

I'm saying that for 90% pilots (even acro pilots) iNav would probably have more than adequate performance. It might not be on the edge, but let's be realistic, how many of quadcopter builders and flyers need this top performance and do racing?

And now people even fly airplanes with iNav.

digitalentity, 2016-03-08:
Please advise what features should we take from CF/BF/RF or what legacy code we should remove from iNav. My suggestions:

Port to iNav:
 1. New mixer code (separate from airmode and make default mixer)
 1. F4 code from @sambas
 1. New resource handling code from @ledvinap (when ready)
 1. SD-card logging (I can forecast new targets with SD sockets popping up all around)

Remove:
 1. Some pre-defined motor mixers (keep only X4, X6, X8, Y6, Tri, Airplane, Flying wing). Maybe limit some targets to even less pre-defined mixes depending on memory and motor/servo output count
 1. emf_avoidance feature - nobody uses it
 1. Some targets incapable of navigation or unused (i.e. CJMCU, STM32F3DISCOVERY, what else?)

More suggestions?

martinbudden, 2016-03-14:
I would discourage getting rid of the CJMCU target. Although, as you say, it is incapable of navigation, I think the discipline of maintaining a 64K ROM build is a good one. It means clean separation of different parts of code is enforced, and helps stop unwanted dependencies creeping in.

I also think the CJMCU is capable of becoming a decent little indoor practice quad. I've built a variant with a sub 32K ROM, so I can certainly see a variant with iNav, sonar based altitude hold (I'd like to try that indoors) and NRF24L01 receiver...

As for what to port, I like your 4 suggestions. I'll think about this a bit more though. I'm not really very familiar with iNav. Do you have the automated build/test system running? If not, that would certainly be useful.

digitalentity, 2016-03-14:
@martinbudden 
Ref CJMCU - I like your sugestion - this target will stay, I think iNav could be stripped down to fit into 64K.
Ref. NRF24L01 - feel free to post a PR here, I like the feature :smiley:
Ref. automated build/test system - no, not at the moment, but I'd like to have one. Suggestions?

lainchir, 2016-03-17:
Hi DigitalEntity. Just found this fork, and as you've seen, I'm having issues with CF1.12. Look forward to trying iNav very soon. Personally, for me... a stable platform with ALT hold, GPS modes and autolevel + LED strips is fine. :) I have oneshot enabled because my ESC's support it, but to be honest... I'm certainly not techy in the programming front, or have understanding of how these boards work, with gyro / acc loops and cycle times... I'd really just like firmware that flies well. 

Thank you so much for all your help with these boards though. This goes for the whole of the team... Hydra, Boris B etc. Without your dedication and time, we'd have very little. Thank you.

Linjieqiang, 2016-03-24:
Port to iNav:
1.New mixer code (separate from airmode and make default mixer)
2.F4 code from @sambas
3.New resource handling code from @ledvinap (when ready)
4.SD-card logging (I can forecast new targets with SD sockets popping up all around)

Thanks for your great work again.I think you can think supporting for F4 Board and save parameter or waypoints or log datas from blackbox in SD-card!

digitalentity, 2016-03-24:
1. New mixer code is ported to FP-PID branch, when it's ready it will make it to INAV.
2. F4 - scheduled (#130)
3. Not ready yet
4. SD-card is being ported as I type this :smile:

Linjieqiang, 2016-03-24:
Also, you can remove unused support target ,because they don't need inav,such as CJMCU ……this can simplify some codes~!

digitalentity, 2016-04-07:
Ok, thanks for feedback, I'm closing this for now

closed 2016-04-07 by digitalentity