iNavFlight/inav #85: Please re-add LED feature to naze32 target

skaman82, 2016-02-17:
Me and most likely many others love the LED support and use it for orientation, warnings and many other things. You've done a incredible job on inav so far, but removing the led support on naze targets was not the best idea I think. I know, this decision was made because of code size, but there are maybe some other functions that can be left out that are not so important as led support. For example: I think most of us can live without the OLED support or support just 16 LEDs instead of 32 if this helps. In this state it is much more harder to fly LOS and I often can't hear the buzzer on my model and have no visual signal for low voltage alarm.

digitalentity, 2016-02-17:
Assigning to 1.1 milestone - possibly will fit with slimmed GPS code.

digitalentity, 2016-04-08:
Fixed

closed 2016-04-08 by digitalentity