iNavFlight/inav #86: Add support for all external Mag orientations

piposa, 2016-02-20:
Align_sensors() has just support for mag orientations if mag is mounted on board. Missing is support if the flight controller board is mounted perpendicular to Mag/GPS board and external mag is used.
Extend align_mag with additional orientations.

digitalentity, 2016-03-13:
That's rather exotic configuration. I'm not sure if it's worth the effort. I'll keep the issue open for the time being to keep track of the idea.

piposa, 2016-03-13:
I you limit your view to copters, I agree it is a quite exotic configuration. 
I use a FLIP mini in flying wing and I have mounted it verticaly in the wing to have easy access to the usb port.
For airplanes the fuselage often allows easier mounting below the cockpit canopy so that the usb port is accessable from the top. Avoids extension cords or additional openings in the fuselage.
I solve that for now by hand modifying the code prior to compile. I agree extending the switch () statement for all possibilities is too much. Right now I lack a better idea, but could it be done as a kind of matrix? The human readable configuration should be done in the configurator... Not exactly sure how....

closed 2018-03-23 by fiam

brainbubblersbest, 2020-02-14:
Can you post a short howto?
eg if i use a BN880 with cw270 mounted compass
What Values do i need to set if the BN880 is mounted with arrow pointing forward.

And now in addition: Tilt it Backwards because it is mounted on a 30 Degree Cam mount.

This Example is seen a few Times and Users are Confused what is to set here.
Thank you.

commit 1a4a3e88 (2018-02-24): Allow aligning the mag to any rotation

Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029
PR #2833 (2018-03-23): Allow aligning the mag to any rotation
Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029
commit 948f2aed (2018-03-23): Allow aligning the mag to any rotation

Introduce 3 new variables which allow setting the decidegrees
for the mag sensor alignment. When any of these 3 variables
are non-zero, mag is assumed to be mounted off-board and
"align_mag" as well as the board alignment are ignored.

Settings are named align_mag_roll, align_mag_pitch and
align_mag_yaw.

Fixes #86
Fixes #1029