iNavFlight/inav #87: Throttle management in GPS-HOLD + ALT-HOLD mode

Mathotg, 2016-02-23:
The throttle management is a bit strange when
1) taking off with GPS-HOLD + ALT-HOLD mode enabled. Full throttle is needed for a while and the machine goes up very slowly. With the time, the throttle management improve and the altitude hold is around 50% of the stick. A video is available on https://www.youtube.com/watch?v=Ucc2pPwoOeA
2) When in flight switching to GPS-HOLD + ALT-HOLD ; the machine drops and if enabled to low, basically you finish on the ground.  here below is the graph showing the logs

[image]

It seems that with set nav_use_midrc_for_althold = ON it's better but I need more testing.

I'm pretty new to INAV and I'm running the original gains. My feeling is that the integral term fro the gain for alt-hold takes too much time to stabilize.

digitalentity, 2016-02-23:
Yes, taking off in altitude hold mode needs improving.

Try upping your vel I and let us know of your findings. Thanks!

Mathotg, 2016-02-24:
Prior the test, I did change to 0.015 the Vel I gain following advise from Konstantin Sharlaimov . The take off is better and the drone "JUMP" to 50 cm eight quickly and then slowly goes up. after around 20 seconds the throttle management become normal. The other thing, I noticed is that at arming, the motors spin quickly for a short period. that's probably the differential gain which is taking effect.

I have to say that I do not know how the code is made for the differential and integral term computation but I suspect that the integral term is fill in with 0 up to the arming and for the differential n-1 signal also. If I'm right, filling in with throttle min value prior arming and throttle value prior inflight switch should decrease the take off issue and the inflight ALT-HOLD swith ON drop

digitalentity, 2016-03-18:
@Mathotg 
Is this issue still present in latest "master"?

Mathotg, 2016-03-18:
@Konstantin,

I did not dare to download the master so far. I was not sure that it could
be downloaded any time without having the risk of having an imcomplete
package. So I'm flying the stable version with the modification for the PID
gains that I did incoporate following the github issue.

Was there any change in the master concerning the arming and throttle
management?
I yes I will download and compile for the SP3racing and test this weekend
or next week depending on the wheather.

Let me know and I will do it

regards

Mathot Gilles (Alias EQ45 on Rcgroups)

digitalentity, 2016-03-18:
Just came to me that this is still an issue - if you switch althold before arming, altitude setpoint does not reset and with baro drift the error grows - this in turn leads to jumps at arming.

Stay tuned for a fix.

Mathotg, 2016-03-18:
Thanks for the info

Is the work around to arm and then switch to alt hold before take off?

Mathot Gilles

digitalentity, 2016-03-18:
Yes, exactly. Please check it if you have the time.

Mathotg, 2016-03-19:
Hello Konstantin,

The work around does not seems to work...
Here his what I tested:

1) connect the battery and wait for satellites
2) arm the motor but no take off
3) switch to pos+alt mode : there the motor spin dangerously for a few
seconds. The quad is nearly taking off himself
4) full throttle to take off

Hope this helps

Regards

Gilles

PR #150 (2016-04-10): NAV: Reset AH/PH position if unarmed
Closes #87
commit ab61b302 (2016-04-10): Merge pull request #150 from iNavFlight/ah_ph_unarmed_reset

Extra safety for ARMing with NAV modes enabled. Closes #87. Closes #153
closed 2016-04-10 by digitalentity