iNavFlight/inav #89: Limited servo throw for fixed wing in Horizon/Acro mode without throttle.

oleost, 2016-03-05:
**Issue:** When in stabilization modes like acro and horizon there is less servo throw when not throttle is engage. Even with airmode enabled.

**Why its and issue: **Many airplanes are suitables as semi gliders, so we throttle up and throttle all the way down and fly without throttle, when doing this we need all the servo movements we got.

Also when landing an airplane its normal to cut throttle and glide in for landing, and also here we need full servo movement to hold the plane steady. With abit of wind I had to put plane in passtrough mode to be able to correct enough.

With limited servo throws blackbox log I do the following:

Arm
Full aileron in passtrough (Getting full aileron throw)
Full aileron in horizon (Getting limited aileron throw)
Full aileron in horzion + throttle. (Getting full aileron throw again)

[CLI dump.txt](https://github.com/iNavFlight/inav/files/159623/CLI.dump.txt)
[limited servo throws black box log.TXT](https://github.com/iNavFlight/inav/files/159624/limited.servo.throws.TXT)

csurf, 2016-03-05:
@oleost,
Although I haven't been able to do a flight-test yet, I just tested my servo throws on the workbench in passthru & stabilized modes (gyro/acro & horizon). The throws seemed equal in both.

I've noticed that, during flight, the stabilized modes always tend to feel a bit dampened, 'dead', and less responsive than when in passthru. However, this is normal, as the stabilizing action of the FC is somewhat 'fighting' with the pilot for control of the servos/control surfaces, and often there's a bit of a conflict. Perhaps this is what you're experiencing? just a thought...

digitalentity, 2016-03-05:
This might be a result of I-term zeroing at zero throttle. Enabling AIR_MODE when in the air might improve the situation. But please avoid enabling AIR_MODE on the ground - it will result in I-term windup and cause trouble with takeoff.

oleost, 2016-03-05:
Hi, I already tried using Air_mode to help on this.

Here is a video showing it:
https://goo.gl/photos/8njL5FfDCemqC9K76

And black box log
[Limited servo throws with video.TXT](https://github.com/iNavFlight/inav/files/159907/Limited.servo.throws.with.video.TXT)

digitalentity, 2016-03-05:
@oleost this is strange. As soon as I get back to my boards, servos and other rc-stuff I'll investigate this further. BTW what PID controller is this?

oleost, 2016-03-05:
@digitalentity Everything is as good as default iNav values with "Airplane" Mix, so its MultiWii Rewrite.

sppnk, 2016-03-19:
Hi @oleost , in your video I see the aileron moving weirdly  just when you start the motor. I have the same behaviour with iNav 1.0.1 in all the servos of my plane. I have reported this in rcgroups too.

oleost, 2016-03-19:
@sppnk I got no wierd servo movement. Just limited throw. Im holding full aileron when applying throttle, thats why the aileron moves at the same time, going from limited to full servo throw.

@digitalentity I figured out something else today, if I set "ROLL rate", "PITCH rate", "YAW rate" in the PID tuning to 1.0 instead of default 0.0 I almost get no limited throw at all without throttle.

sppnk, 2016-03-20:
@oleost, my plane is horizontal an?d calibrated. this is annoying. Do you have the same version of iNav?

digitalentity, 2016-03-20:
@oleost, @sppnk 
Ok, now we are getting somewhere. When R/P rate is set to 0, angular rotation target is limited to 200 degrees/second at full stick deflection. PID is working to satisfy this rate, if servo throw is not sufficient to satisfy this rate, R/P P-term needs to be higher. 
So this "limited" servo throw is meerly a matter of proper R/P PID tuning.

sppnk, 2016-03-20:
@oleost could you try not to touch the aileron sticks and push only the throttle when armed? In my case if I do this, all the servos move to one side, not only ailerons. Please report. Thx.

oleost, 2016-03-20:
@digitalentity But why the difference with throttle or no throttle.

Important to remember with airplanes is that even with no throttle the airplane can be at high speed ( example in a dive )

oleost, 2016-03-20:
@sppnk no movements if applying only throttle

sppnk, 2016-03-20:
@oleost Just the throttle channel moves in the receiver tab. Do you use iNav 1.0.1?

oleost, 2016-03-20:
Yes and no, I started with iNav1.0.1 and moved on to a self compiled
version of Master branch. But I'm on f3 target and not cc3d
20. mar. 2016 8.44 p.m.

sppnk, 2016-03-20:
well, it seems my problem has to do with 6 steps acc calibration. I will try to do this and report.

What I see also is that these high "compensations" in the servos dissapear if Throttle goes below MinThrottle parameter, (i.e. cutting engine). I think is the same thing you are talking about, @oleost .

I think that with motor off we should have the same PIDs working as with motor on. For example to glide.

digitalentity, 2016-03-20:
I was unable to confirm the issue. Servos work as expected in all modes.
1) Passthrough = direct stick input applied to servos
2) Acro + low throttle = PID active, only P-term is applied to servo
3) Acro + high throttle = PID active, I-term adds more servo throw over time
4) Angle mode + low throttle = PID active, R/P P-term and LEVEL P are applied to servos
5) Angle mode + high throttle = PID active, R/P P&I-terms and LEVEL P are applied to servos

Limited servo deflection in ACRO mode - matter of proper ROLL/PITCH PID tuning, servo deflection must be able to sustain the requested angular rotation rate.

Servo drift in ANGLE mode and throttle above min-check - I-term in action. Put the board perfectly level and you won't see it.

digitalentity, 2016-03-20:
AIR mode does it job in allowing I-term only when ARMED, checked and works as expected.

sppnk, 2016-03-21:
@digitalentity in my tests, if I cut throttle, the servos inmediatly reduce their throw (in angle and horizon modes), exactly like @oleost says in his first post.

digitalentity, 2016-03-21:
You have different throw at low throttle and at high throttle due to I-term limiting at zero throttle. And I-term is winding up **increasing** your servo throw only because the airplane is unable to satiafy the command (https://github.com/iNavFlight/inav/issues/26#issuecomment-199069450)

sppnk, 2016-03-21:
Thx. I will test this in the air probably this week.

digitalentity, 2016-03-21:
Good luck! You may also want to increase ROLL/PITCH P if response to the right stick in flight is not sharp enough

oleost, 2016-04-03:
Closing this one.

sppnk, 2016-05-15:
This issue still remains. With zero throttle, the plane's control surfaces are much less maniobrable. I guess we should have the same control with or w/o motor.

It is a problem usually for landing, as I have to increase trottle to get more pitch down, instead of cut motor. It is also a problem if we want to glide with nice control when not in 'passthrough'

digitalentity, 2016-05-15:
@oleost can you test 3e6204d? Should fix the issue.

oleost, 2016-05-16:
Sure, will try it when I get a chance.

oleost, 2016-05-16:
Sure I can test.

But lets recap:

With airmode enable before, there was no I-term attenuation? It was only an interlock so after arming one had to go above a certain % of throttle to get rid of the I-term attenuation?
So with airmode one had to throttle up, and then go to zero throttle and test, and there would be no difference in servo movement?

I aggre that "airmode" should be default "on" on airplanes.

But with new code there is no interlock right? So I-term will always wind up after powering up?

digitalentity, 2016-05-16:
No, with new code I-term will wind-up only if AIR-MODE is disabled. If AIR MODE is enabled I-term is only allowed to shrink, not grow.

sppnk, 2016-05-16:
Then, If I understand well, enabling AIRMODE solves this, isn't it?

oleost, 2016-05-16:
Okey, will test.

One fear is that I term will grow while just standing on ground  getting ready to fly.

One solution would be that I-tern is not allowed to grow while unarmed and throttle has not been over 30%. After throttle been over 30% it's allowed to grow below 30% aswell. Resets when it's disarmed. What you think?

oleost, 2016-05-17:
@digitalentity  tested and confirm that its no difference with throttle on/off now.

Possible to get fw-improvements updated with current master?

sppnk, 2016-05-17:
great !

oleost, 2016-05-28:
@digitalentity One still need airmode to get full deflection on fixed-wing. Just tested again with new set of PIDS with not enough P gain to allow full servo movement on P itself.

Airmode OFF = Not full servo throw when no throttle (But armed)
Airmode On = Full servo throw with no throttle.

digitalentity, 2016-05-28:
Yes, that's the way it is. At zero throttle with AIR MODE off I-term is
inactive and it's the I-term that allows full servo throw when P is not
enough.

oleost, 2016-05-28:
@digitalentity Sounds good, just notice you sad "No, with new code I-term will wind-up only if AIR-MODE is disabled. If AIR MODE is enabled I-term is only allowed to shrink, not grow."

digitalentity, 2016-05-28:
I guess I was wrong when typing this :)
It's like this at zero throttle:
1) AIR MODE disabled: I-term reset and inactive
2) AIR MODE enabled: I-term only allowed to shrink when right stick
centered, I-term is fully active when stick in deflected.

oleost, 2016-05-28:
Cloosing this then?

digitalentity, 2016-05-28:
I think so. Just need an update to the wiki that airplanes should use air
mode to ensure proper zero throttle operation.

oleost, 2016-05-28:
Added it to Step 5 in fixed wing guide. https://github.com/iNavFlight/inav/wiki/3.-Fixed-wing-guide

odixxx, 2016-05-29:
AIR MODE why not turn off the engine?
how to plan a landing?

oleost, 2016-05-29:
@odixxx You are saying when turning on airmode the motor is running even with zero throttle?
This is not normal, and should not happend.

But in order for us to look at whats wrong, please provide more infomation like CLI dump and a blackbox log.

odixxx, 2016-05-29:
yes all right, the engine is running
I can write the blackbox

odixxx, 2016-05-29:
[blackbox_log_2016-05-30_000840.TXT](https://github.com/iNavFlight/inav/files/288551/blackbox_log_2016-05-30_000840.TXT)

oleost, 2016-05-29:
And cli dump? ...

odixxx, 2016-05-29:
[cli dump.txt](https://github.com/iNavFlight/inav/files/288556/cli.dump.txt)

oleost, 2016-05-29:
Can you describe what you did in that blackbox log?

Are you armed the entire time?

Because for what I can see it only at min_throttle 1150, except when the motors are not running its at 1000.

min_throttle	These are min/max values (in us) that are sent to esc when armed. Defaults of 1150/1850 are OK for everyone, for use with AfroESC, they could be set to 1064/1864.

So I guess either calibrate your ESC or lower the min_throttle settings.

odixxx, 2016-05-29:
blackbox  AIR MODE turning on time 00:11.750

oleost, 2016-05-29:
Yes I can see.

But output only shows 1150us, and its 1150us you have set as min_throttle when armed.

Before you turn on airmode, it only 1000us. Not sure why output is only 1000us before activation, or is it not armed?

Anyway, I think if you calibrate your ESC, or lower min throttle it will be fine.

odixxx, 2016-05-29:
All this time he was armed
1150us on AIR MODE?

odixxx, 2016-05-29:
[blackbox_log_2016-05-30_011726.TXT](https://github.com/iNavFlight/inav/files/288574/blackbox_log_2016-05-30_011726.TXT)
 00:07.811
changed min 1160

oleost, 2016-05-29:
Just lower your min_throttle to 1000 and you should be good.

odixxx, 2016-05-29:
changed min 1064
It does not start,
Thank you

oleost, 2016-05-29:
@digitalentity So whats expected behaviour here?

According to documentation min_throttle is sent when the aircraft is armed.

min_throttle	These are min/max values (in us) that are sent to esc when armed. Defaults of 1150/1850 are OK for everyone, for use with AfroESC, they could be set to 1064/1864.

Looking at the blackbox log only 1000us is sent while being armed, even with min_throttle set to 1150. But when going in airmode it goes to 1150.

What you think?

sppnk, 2016-05-30:
Hi, I am using version 1.1 and I tried yesterday AIRMODE because of the plane behaves very diferent with and without throttle applyed (it is very hard to land).

But the motor starts when arming and doesnt stops never (only when dissarming). Is it the same issue like @odixxx or have I to wait to next version? (I dont know exactly if you are talking about 1.1 or last master version).

oleost, 2016-05-30:
Probably  because it goes to min_throttle. You can simply lower it to the point where your motors are stopped.

For a more permanently fix I guess we either need go down to mincommand even in air mode with fixedwing, or simply have full pids without air mode.

oleost, 2016-05-30:
Not home, so cant test.

But would this bd6a17f03d7424991f6ff200ce8800ad090e12c0 change so we dont need to enable airmode to get full pids at low throttle?

sppnk, 2016-05-31:
I guess it should do it, so no neccessary enable airmode. But I have not tested yet.

oleost, 2016-06-23:
@digitalentity This issue could be closed, but for it to be perfect it would be when airmode is enabled and state = fixedwing it does go to zero throttle aswell, so we dont have to lower min_throttle.

digitalentity, 2016-06-24:
Closing. Free free to re-open if necessary

closed 2016-06-24 by digitalentity

sppnk, 2016-06-25:
tested two planes yesterday with fw-improvements branch merged to latest master. It works without airmode enabled.