iNavFlight/inav #90: FW - add crosstrack error support

csurf, 2016-03-05:
more info:
http://www.rcgroups.com/forums/showpost.php?p=10263920&postcount=12
http://diydrones.com/forum/topics/anyone-knows-how-to-measure-the-lateral-crosstrack-error

digitalentity, 2016-03-05:
Good idea, thanks

digitalentity, 2016-06-22:
This won't happen in 1.2 sorry.

oleost, 2016-09-30:
@digitalentity Postpone to 2.0 ?

digitalentity, 2016-09-30:
I doubt this will happen any time soon. This is really low priority :smile:

breadoven, 2022-11-26:
This is effectively dealt with by https://github.com/iNavFlight/inav/pull/8234 so this can be closed. Could rename `distToCourseLine  `parameter to "cross track error" since this is what it is in fact. Could also be made available as a `posControl `parameter although I'm not sure what use it has other then for display on the OSD perhaps.

closed 2023-10-27 by DzikuVx