iNavFlight/inav #91: NAV - emergency landing for FW

csurf, 2016-03-05:
request for comments on how an emergency landing routine should be setup for fixed-wing models. 

My thoughts are that the plane:
- should descend downward in a light, banking circle, with slight downward pitch 
- rate/radius of circular descent could be tracked by monitoring heading values from MAG sensor.
- Throttle should be kept at a minimum, user-configurable value 
- when the plane has reached some pre-defined minimum altitude, throttle should be killed and plane allowed to perform a level glide to earth. 
- Optionally, disarm the plane after some landing detection timeout.

feedback is welcome...

oleost, 2016-04-05:
Sounds good. Only issue is that not every plane have MAG sensor. So it should probably be able to do blind circle aswell.

sppnk, 2016-04-05:
@csurf, It looks a good routine for emergency landing. Should be careful with landing detection or remain it optional, to not disarm the plane in flight. Also check that some FC dont have MAG, as @oleost says.

oleost, 2016-04-05:
I would say never disarm, only throttle cut. As disarming would prevent
user from arming again.

digitalentity, 2016-07-13:
Emergency landing is a safety feature if navigation is not possible. It is invoked when sensors can't be trusted, so FW should be able to land blindly.

So far initiating a constant yaw rotation and a slight downward pitch with minimum throttle looks like the best way to do it.

sppnk, 2016-07-13:
@digitalentity, AFAIK the automatic GPS assisted modes act mainly over ailerons, not rudder (it would be good to have some parameter to give more weight -a percent-to rudder action than ailerons action). 

So, if GPS fails, the rudder turns could give slower landing, I guess. Or a 70% rudder 30% aileron or something like that.

digitalentity, 2016-07-13:
Latest code should use both rudder and ailerons to do a faster turn. If you can test latest "master" branch and send me a blackbox log of RTH it would be perfect to verify that both rudder and ailerons are used.

sppnk, 2016-07-13:
Then, that feature came when 'fw_improvements' was merged? If so, I have two planes with it, but then, is there a parameter to distribute the turn action bettween ailerons and rudder or is it hardcoded?

digitalentity, 2016-07-13:
@sppnk yes it was a feature merged with fw-improvements. Yaw amount is calculated automatically based on [physics of banked turn](https://en.wikipedia.org/wiki/Banked_turn). In a nutshell at smaller velocities it adds more rudder and at large velocities - less rudder.

oleost, 2016-07-13:
No throttle please,  or at least user configurable.

When a plane emergency lands I don't want any big propeller spinning,  for
all you know it can hit someone.

Turning with a predefined roll and optimal glide angle would probably be
the best.

Which values that would be universal "safe" glide angle for most rc
airplanes I don't know.

@sppnk <https://github.com/sppnk> yes it was a feature merged with
fw-improvements. Yaw amount is calculated automatically based on physics of
banked turn <https://en.wikipedia.org/wiki/Banked_turn>. In a nutshell at
smaller velocities it adds more rudder and at large velocities - less
rudder.

—
<https://github.com/iNavFlight/inav/issues/91#issuecomment-232320662>, or mute
the thread

digitalentity, 2016-07-13:
@oleost sounds reasonable to me. Configurable glide and bank angles, yaw rate and throttle would be parameters for emerg. landing. Small glide angle will keep the airplane from stalling even at zero throttle.

I'll need advise on how you'd prefer to land a plane (glide angle and yaw rate/bank angle).

oleost, 2016-07-13:
TBH I'm not that experienced fixed winger. I could always find option angle
for my plane, but since it's a trainer it will vary a lot depending on
airplane/wing design.

But I can try and hear other places. And report back.
13. jul. 2016 14.53 skrev "Konstantin Sharlaimov" <notifications@github.com

oleost, 2016-07-26:
@digitalentity Tried searching around. But only got answers like get it to go into a flatspin and crash. 

But since this is from people that have been flying airplanes for 30 years, and not used to the wonders of flight controllers I think we still will be better off by doing as we talked about.

https://drive.google.com/open?id=0B-yHzNtOXxbEWGRObmYwbkpXWXc

digitalentity, 2016-07-27:
@oleost at the moment pre-configured yaw/roll/pitch is probably the most viable option for emergency descent. As for normal descent - just continue loiter around home point and decrease altitude gradually and slowly (pretty much the same logic we currently use for MRs)

oleost, 2016-07-27:
Hello,
In Failsafe RTH with GPS.
When the plane arrived home close engine and circle down .
Stages
1 / Enable posHold / Lotier and change the target Altitude to 5m and disarm .
2 / Change target to Altitude 0m.

without GPS
1 / Run engine at Failsafe throttle ~ 20 sec and apply ~ 5 degrees left roll to do a blind circle back to catch the signal .
2 / Disarm , continue circle and apply ~ 5 degree dive to stop runaway .

Options to step 2 if it is windy is to take down the plane really quickly it can drift quite far at high altitude.
If you add it up in the tough role of > 45 degrees and little dive then it will come down relatively quickly.

Happy Holidays..
Patrik

oleost, 2016-07-27:
Reply I got form PatrikE. (Used google translate from Swedish to English)

> Hello,
> In Failsafe RTH with GPS.
> When the plane arrived home close engine and circle down .
> Stages
> 1 / Enable posHold / Lotier and change the target Altitude to 5m and disarm .
> 2 / Change target to Altitude 0m.
> 
> without GPS
> 1 / Run engine at Failsafe throttle ~ 20 sec and apply ~ 5 degrees left roll to do a blind circle back to catch the signal .
> 2 / Disarm , continue circle and apply ~ 5 degree dive to stop runaway .
> 
> Options to step 2 if it is windy is to take down the plane really quickly it can drift quite far at high altitude.
> If you do an tough roll of > 45 degrees and little dive then it will come down relatively quickly.
> 
> Happy Holidays..
> Patrik

csurf, 2016-07-27:
just to add to throttle-cut discussion, for boards equipped with baro, it would be cool to optionally add a configurable height to cut throttle during landing. So, for example, you can set it to a height of aprox 1-2m, and it will kill the throttle when the FC detects that the plane has descended to this height during landing. This way, you can still glide down with low throttle to avoid a stall (good for windy conditions). 

However, if there's no baro, or if baro fails for some reason, then simply override this option and kill the throttle for entirety of emergency landing.

oleost, 2016-09-30:
@digitalentity This would be great to get some sort of implementation of this to 1.3

As it can be scary without emergency land for FW users.

Maybe just adopt PatrikE protocol?

In Failsafe RTH with GPS.
When the plane arrived home close engine and circle down .
Stages
1 / Enable posHold / Lotier and change the target Altitude to 5m and cut motors.
2 / Change target to Altitude 0m.

without GPS
1 / Run engine at Failsafe throttle ~ 20 sec and apply ~ 5 degrees left roll to do a blind circle back to catch the signal .
2 / cut motors, continue circle and apply ~ 5 degree dive to stop runaway .

Just as an starting point for the amount of roll we can use what values my plane have when doing loiter above home.

[image]

Link to the blackbox log when doing RTH loiter on log 4/4 around same timestamp as screenshoot https://drive.google.com/drive/u/0/folders/0B-yHzNtOXxbEU1BiSnFacGtBTmM

digitalentity, 2016-09-30:
As a first stage this will go as an "emergency landing" protocol (is enabled when in PH/RTH and GPS is lost):
1. Apply a predefined roll & run engines at failsafe throttle for defined time
2. Continue roll and cut engines.

What do you think?

digitalentity, 2016-09-30:
"Emergency landing" protocol is not a R/C failsafe, it's a GPS loss failsafe when it's unsafe to continue a GPS mission.

oleost, 2016-09-30:
All this is easy to get confused..

"Emergency landing"
Your plan sound ok, I would also start acending in first stage. As the longer it flies in blind the more drift.

So
1. Apply a predefined roll and dive & run engines at failsafe throttle for defined time
2. Continue roll, accend/dive and cut engine

And for the regular "landing" after and RTH / "Failsafe RTH"
1 Enable posHold and Althold and change the target Altitude to 5m and cut motors.

oleost, 2016-09-30:
And just to verify, lets say you out flying, you loose RC link, and you also lost GPS, it will simply to go emergency land because of it tries to enable RTH without an valid GPS lock?

martinbudden, 2016-10-05:
@digitalentity , how far advanced is this? Unless it is really almost finished, I would defer it to 1.4.

digitalentity, 2016-10-05:
Just an idea, no coding done yet.

digitalentity, 2016-11-17:
Fix pending by #799, closing.

closed 2016-11-17 by digitalentity