iNavFlight/inav #92: NAV - cruise mode for FW

csurf, 2016-03-05:
cruise mode is somewhat of an extension of POSHOLD for fixed-wing models. The purpose is to allow for simple, straight-line 'point & go' flying, where FC takes care of maintaining stability, altitude, and heading, allowing user to relax and handle other tasks. Good for long-range flights.

When activated, what it does is:
- auto-enable ANGLE mode  in order to maintain level flight
- auto-enable fixed-wing ALT mode and MAG/heading-hold modes
- set a distant way point (~1km away) and fly toward it
- maintain heading & altitude during waypoint flight
- periodically auto-update target waypoint to new distant point (~1km) based on current heading
- yaw/roll (RUD/AIL) inputs should temporarily release auto-flight & update target waypoint based on new, user-defined heading after sticks are released. (*altitude control with AIL input might be tricky as plane will tend to descend when rolling... not sure, feedback required)
- pitch/ELE input should temporarily disable ALT hold and allow user to climb/descend. New altitude should be set & held after ELE stick is released
- *auto-throttle control seems like it might be tricky in order to maintain both (ground) speed & altitude. Easiest method would be to just ignore & not handle throttle control, and force pilot to manually manage throttle. Otherwise, more complex code would be required. Correct? 

feedback is required...

brat002, 2016-03-06:
I am dreaming about this feature!

digitalentity, 2016-03-06:
Same logic can be applied to multicopters as well. It would be a great feature to have for FPV flights. Scheduling to 1.2 milestone.

sppnk, 2016-07-25:
I am doing many flights with my new fpv gear. Inav rocks !!.

As all the main stabilization and gps modes are working now, I think it is time to take in consideration other interesting modes. I would like to have a cruise mode working, and a optimal-battery-saving cruise mode.

In the meanwhile, does "heading lock" mode work for airplanes with gps (no mag)?

oleost, 2016-09-30:
@digitalentity Postpone to 2.0 ?

FPVZaphod, 2017-09-04:
This would be great. Maybe also for Flying Wings without Rudder?

trailx, 2018-02-02:
This enhancement would put inav into a category with vector and pixhawk.

@FPVZaphod Agreed, would need to work with flying wings.  Alternative is to possibly mix rudder ouput into aileron outputs in smix to spoof it to work?

IVData, 2018-09-12:
@digitalentity I think this can be closed due to cruise mode in 2.0.

shellixyz, 2018-11-19:
@IVData For now cruise mode is only working for fixed wing

closed 2023-10-27 by DzikuVx