iNavFlight/inav #93: Throttle compensated mag calibration

theArchLadder, 2016-03-06:
I'm still using my internal mag on my 250 racer quad. The FC is right above the PDB, this gave me up to 30 degrees error on compass reading with violent toiletbowling.

I have solved this by looking at a few blackbox logs and checking what the MAG values are when i arm, and then what the values are right at takeoff.  I then modified magzero_x,y,z for that. This makes my compass up to 30 degrees wrong before takeoff and wrong on high throttle, but correct at hover throttle. This has been very successfull, i never have any toilet bowling now, and iNav seems really happy with the compass reading as far as i can tell.

This is one example of what mag readings from blackbox looked like for me:
`Idle, hover, difference`
`-107  -159  47`
`-319  -453  188`
`-677  -560  -157`

This is my compass calibration before doing anything:
`set magzero_x = -12`
`set magzero_y = -72`
`set magzero_z = 439`

And this is what i ended up with after compensating for throttle:
`set magzero_x = -146`
`set magzero_y = -20`
`set magzero_z = 556`

This is not user friendly though, on my SPRacingF3, the compass is rotated, and magzero_ are applied before rotating, so for me magzero_y and magzero_x are swapped, and magzero_y is inversed. It would also be better to have this offset scaled with throttle, so compass reading can be accurate at both idle and hover, and high throttle values. I think a linear scaling should be good enough for a start since my quad flies great even without scaling.

So here is my feature request:
- Create 3 new CLI settings called something like magzero_hovercomp_x,y,z.
- Scale these values with the help of throttle, i think the formula should look something like this:
`scaling = (throttle - throttle_min) / (nav_mc_hover_thr - throttle_min)`
So that the scaling is 1.0 at hover, since this is the most important throttle position and where we get the difference from the blackbox log.
- Add these values to mag readings after the mag has been rotated.

I think this could help many quads, even those who have external mag on small quads.
Sadly, my programming skills are not good enough for me to do this myself.

digitalentity, 2016-03-06:
Good thing to have, thanks for investigating

DzikuVx, 2016-03-16:
I think it would be a hell to calibrate this feature for anything else than hover and thus not usable for example at 3/4 of throttle. 

There can also be a case when only one of power wires interferes with magnetometer. For example power wire to ESC motor 1. So, scaling would be usable only when this one motor is getting enough current to generate mag field strong enough to alter magnetometer readout.

digitalentity, 2016-03-16:
@DzikuVx 
Throttle-compensated mag calibration is not a magic tool, but in many cases it can make mag working - in many cases interference is big enough to confuse navigation, but not big enough to make compass readings unusable.
Still, not a top-priority task.

theArchLadder, 2016-04-10:
I got this to work now, i have been flying with it for a few days and it seems to work really well. I never see any toilet bowling! Before, when i put this compensation in magzero, i got up to 30deg error with motors off, and i had to hover for about 30 seconds before the compass reading got accurate, now it just works! Poshold, RTH, WP missions etc works great.

This is the code i'm using in compass.c
```
float thrComp = 0;
if (ARMING_FLAG(ARMED)) {
    thrComp = (float)(rcCommand[THROTTLE]-1000)/(1300-1000);
}
magADC[X] -= magZero->raw[X] + magZeroHoverComp->raw[X] * thrComp;
magADC[Y] -= magZero->raw[Y] + magZeroHoverComp->raw[Y] * thrComp;
magADC[Z] -= magZero->raw[Z] + magZeroHoverComp->raw[Z] * thrComp;
```
I also moved `alignSensors(magADC, magADC, magAlign);` from below to above this code.

The hardcoded values needs to be min_throttle and nav_mc_hover_thr, but i have not figured out how to get those yet.

I looked at a blackbox log and compared mag values at idle and at hover, with my upgraded quad, x and y is both about 170 units wrong for me, z have a huge error of 640, but it still seems to work nice when flying now.

What @DzikuVx said is true, but during navigation stuff, the throttle is always close to hover and there is no big difference in speed between the motors for most of the time. Heading relies mostly on the gyro anyway, so even with a few random noisy readings from the mag, the heading will be accurate as long as the average mag reading is right.
The error scales very well with throttle, so only calibrating at hover throttle should be good enough.

closed 2018-05-27 by stale[bot]

robustini, 2022-09-02:
> @DzikuVx Throttle-compensated mag calibration is not a magic tool, but in many cases it can make mag working - in many cases interference is big enough to confuse navigation, but not big enough to make compass readings unusable. Still, not a top-priority task.

In Ardupilot this feature called [CompassMot](https://ardupilot.org/copter/docs/common-compass-setup-advanced.html) has existed for many years and is convenient in those cases where EMI is impossible to eliminate.
And it works very well, the deviation is automatically calibrated relative to the throttle or current sensor if available.
It would be interesting to have it in iNav too.