iNavFlight/inav #94: nav_posr_d max value needs to be 10 times stronger.

theArchLadder, 2016-03-06:
I have used a custom iNav build that let me set nav_posr_d 10 times higher than default, this has improved the behaviour of my quad, see this RCGroups post in the iNav thread for more info:
http://www.rcgroups.com/forums/showthread.php?t=2495732&page=251#post34180060

Weather has been bad so i have only been able to do a few test flights, values between 75-150 (or 750-1500 with the old scale) seemed to work well on my quad and it's always nice to be able to play with even higher values, so i think 10x scaling is about right based on my limited test flights. Nobody seems to use posr_d right now so rescaling it should not cause angry users.

I just changed this line from 1000.0f to 100.0f, not sure if this is the right place to adjust and i'm new to github and have some reading up to do. So i will not do a pull request this time.
https://github.com/iNavFlight/inav/blob/master/src/main/flight/navigation_rewrite.c#L2132

digitalentity, 2016-03-06:
Will make the change in a day or two

digitalentity, 2016-03-06:
@theArchLadder 
It is done. Please test it and try to find a suitable POSR_D default value (currently at zero).

commit 621c8e3b (2016-03-06): NAV: Change D-term scaling for NAV PIDs. Closes #94
closed 2016-03-06 by digitalentity