iNavFlight/inav #99: gcc 5.3 generates unsafe code

stronnag, 2016-03-07:
I've been using arm-gcc 5.3 ever since it became the Arch Linux default about a month ago. Since then I've seen:

- Inexplicable failure to perform PH (instead shooting off in a random direction). After a soft FC reboot under the same external conditions (GPS, wind), the PH might then execute perfectly.

- Last Friday, mega roll of death during otherwise controlled RTH autoland (rotating at multiple rev /sec), uncontrollable crash.

- Failure of yaw to be recognised with DigitalEntity's new pid controller.

Rebuilding with arm-gcc-4.9 resulted in the new pidc working correctly. I suspect it will also fix the intermittent PH issue.

I do not recommending building inav with gcc 5.x at this stage, in particular, Arch Linux users should downgrade to 4.9.

digitalentity, 2016-03-07:
Thanks for investigating this. I've also encountered gcc-5.x related issues, mostly with USB VCP on CC3D board. I will avoid using gcc-5.x for building iNav until this is investigated furhter.

theArchLadder, 2016-03-09:
Thanks for reporting.
I have been using a arm-gcc 5.3 on archlinux for a few days and have done more than 10 flights, i have not seen any issues yet though.
I guess i will use 4.9 in the future just to stay safe.

MauroMombelli, 2016-03-09:
i think problem like that can be solved in a clever "continuous testing" way: we tag a specific version, you run it on a real case scenario recording all the sensor input and output.

Then a sort of "replay" function where the board read that logged data instead of sensors and check the results. If inconsistencies are detected, then are recorded and human will have to analyze.

This will not be possible with many board (lack of space or reading speed for example) but one should be enough to start with.

Then if more board will come out with that capability, that recording could be enabled by default and when crash happen hopefully a *reproducible* study case will be available

stronnag, 2016-03-11:
I flew 4.9 hex (fp-pid) and it was so much more stable than anything I've flown on 5.3 for the past 6 weeks or so. Solid PH/RTH/WP, density softserial telemetry, great GPS data. Damn you 5.3, you own me so many propellers.

digitalentity, 2016-03-12:
@stronnag 
We'll stick to 4.9 then. BTW which board are you testing with? I wonder if it's the float-point emulation that is broken in 5.3?

digitalentity, 2016-03-12:
@MauroMombelli 
Can you elaborate? You suggest feeding blackbox data into the FC instead of real sensor input?

stronnag, 2016-03-12:
@digitalentity, problem on both F1 and F3.

MauroMombelli, 2016-03-14:
+digitalentity exactly, if we know that in a determinate code revision from determinate input we should get a determinate output, if we see discrepancies we know about some compilation/linking/whatever error.
This may seems excessive, but nowadays we get Clang and GCC making a lot of release per years, and this as caused a lot of troubles (i'm thinking about LTO for example). Also would be interesting if GCC and Clang produce different result..
This could also be extended to new code revision; of course things may be different because legit edit in code, so that would need a bit of manual review, but may prevent code with collateral effect breaking.

closed 2016-07-09 by digitalentity