iNavFlight/inav #103: FollowMe mode

eziosoft, 2016-03-08:
Could you add a separate followMe mode to avoid confusion. At the moment FC has to be in Position Hold to follow which is not very good. Maybe there is a simple way to assign PH to only for position hold (which means just to ignore setting the WP255). And add a FollowMe mode which will be doing exactly the same as PH now.

dkisselev, 2016-03-08:
If follow me is basically just PH, does the Quad need to know/manage the fact that it's in an FM state? That seems like a client-side configuration/system, until here needs to be some specific behavior on the quad side that's unique to FM.

Heading control/WP255 could just be rolled into standard PH, couldn't it? If a heading was sent with the position vector, turn that way, otherwise just maintain heading.

eziosoft, 2016-03-08:
The followMe mode will give the user the possibility to enable/disable it from TX.
At the moment when followme positions are sent from the ground station, classic position hold is not available. The user may want to have PH ie in panic situation.

digitalentity, 2016-03-08:
@eziosoft 
Done, see fbeff0a. Featuring new "GCS NAV" mode. It has to be enabled together with POSHOLD by TX AUX to allow PH position to be adjusted by sending WP#255.

eziosoft, 2016-03-08:
Wow you are fast. Thanks a ton.
I will test it asap.

closed 2016-03-08 by eziosoft

iforce2d, 2016-06-22:
Hey @eziosoft did you ever try setting the heading of WP255? It seems to be ignored for me (lat, lon, alt work ok though).
I'm pretty sure it is being set correctly but then must be getting overwritten later, but I've looked through the source code and can't figure out where the overwrite could be....

eziosoft, 2016-06-22:
Hi. You have to have MAG mode also enabled. Only then heading is working.
Bart

iforce2d, 2016-07-28:
Sorry, I forgot to say thanks for the tip. It works nicely with MAG enabled.

oleost, 2016-11-25:
@eziosoft Please have a look at this: https://github.com/iNavFlight/inav/wiki/8.-Navigation-modes#gcs_nav---ground-control-station

This is correct right? Only NAV POSHOLD is requirred, and if you want to change heading you need MAG aswell.

Althold is not requirred?

eziosoft, 2016-11-25:
GCS_NAV should be treated more like permission. When it is ON the app can change position hold coordinates and the altitude. If it is OFF requests to change position hold location and altitude are ignored by the flight controller. It was introduced as an extra safety feature. 
To allow full control by the app (for follow me, etc) position hold, altitude hold, mag and GCS_NAV should be enabled.
If you disable MAG the app can't control heading. If you disable alt hold the app can't control altitude, you will have to do it manually.

Maybe this behavior should be changed as at the current state it is not very intuitive.

oleost, 2016-11-25:
It may be useful to have it this way, since it allows people to customize the behavior. Example follow something while manually yawing, and controlling height for smoother operation.

But I think it maybe should force angle mode as many minimum.