iNavFlight/inav #104: Heading control for FollowMe  (WP255)

eziosoft, 2016-03-08:
Could one of the parameters in WP255 be used as a heading control ?
Also it will be very useful if one particular value outside the heading range can be used to keep last heading.

digitalentity, 2016-03-08:
@eziosoft 
Please see the latest master. You now have the ability to set heading by passing P1 parameter with WP255.

Poshold now executes heading controller, but does not force MAG mode to be enabled. If you want P1 parameter to actually turn the quad you have to enable MAG mode explicitly.

P1<=0 and P1>=360 are considered "keep last heading".

You won't be able to set heading "due north" but 1 deg error is negligible.

eziosoft, 2016-03-08:
Thanks ! That was fast.

closed 2016-03-08 by eziosoft

digitalentity, 2016-03-08:
@eziosoft I didn't test the code. Please check if its working as expected. If anything is wrong please reopen the issue.