iNavFlight/inav #106: Force-arm without gps fix

theArchLadder, 2016-03-09:
I dont always fly with a clear sky above my quad, sometime it could be in a forest, or even indoors.
I know there is a CLI option for this, but imho it would be even better if we could have some way of force-arming without gps.
This way we can both have the cake and eat it. We will not arm without gps by accident, but we can still arm without gps if we really need to and dont have access to the CLI.

The way to do a force arm could be something like this:
Try to arm 5 times within 5 seconds to arm without gps.
This should work both with the stick and switch arming.

digitalentity, 2016-03-12:
Any thoughs on the proposed change, anyone?

oleost, 2016-03-12:
Sounds good.

It's handy to have a backup way to get it flying without GPS. And I doubt
anyone tries to arm 5 or more times within 5s on accident.
12. mar. 2016 10.07 skrev "Konstantin Sharlaimov" <notifications@github.com

DzikuVx, 2016-03-16:
I'm against this change. Probably nobody would use it in real live. Plus, it would affect GPS assisted modes with extra flags required not to enable RTH modes even after lock has been acquired, but protection was overridden. Safety precautions works only when they are simple

digitalentity, 2016-03-16:
I tend to agree with @DzikuVx - people usually don't fly GPS-equiped drones in GPS-denied environments. And for the rare case this should be allowed we have **nav_extra_arming_safety** setting.
If one is flying with nav_extra_arming_safety=OFF he/she is probably smart enough to understand the risks of flying without home position set.
Having the way of overriding safety protocol is equal to having no safety protocol at all - why bother implementing an override, just disable nav_extra_arming_safety alltogether.

closed 2016-03-16 by digitalentity

StNekroman, 2023-05-31:
"people usually don't fly GPS-equiped drones in GPS-denied environments."
Ok, so solution is - to cut out GPS module before flight in such environement, right?

MrD-RC, 2023-05-31:
Why are you bringing up a 7 year old issue? If you want to ignore the GPS and arm. You can bypass the safety checks by yawing right when you arm. 

**set nav_extra_arming_safety = allow_bypass**

Comparing with BetaFlight is wrong. It is not a navigation platform.

StNekroman, 2023-05-31:
> Why are you bringing up a 7 year old issue? If you want to ignore the GPS and arm. You can bypass the safety checks by yawing right when you arm.
> 
> **set nav_extra_arming_safety = allow_bypass**
> 
> Comparing with BetaFlight is wrong. It is not a navigation platform.

Wasn't looking how old this ticket is.
Yes, already found magic CLI command and after spending another 20 minutes found how to actually bypass arm checks via magic with sticks.
But anyway, I still think that having normal checkbox in UI will be clearer for users.

MrD-RC, 2023-05-31:
This really shouldn’t be a UI option. Allow_bypass is the default setting. The actual bypass should only be used on rare occasions, like bench testing or flying indoors. There would be no need or reason to permanently disable these checks; which would require a UI option.