iNavFlight/inav #107: PH question

bk79, 2016-03-10:
@digitalentity  If during PH I move the sticks will the gps coordinate calculated back? or it still keeps the  coordinates of when I activated PH? Example...I'm already with PH activated I move the stick to move it 20-30m ahead, I have a certain speed so when I release the stick it will still have some inertia and go forward. If instead when I release the stick the new coordinates are takes it will it brake and get back to the position of when I set the sticks in middle position. Does it works this way or it will simply get back to the original coordinates of position hold activation?

digitalentity, 2016-03-10:
When you release the stick iNav will calculate new position according to
configured braking (a little bit ahead in the direction of movement to
allow graceful deceleration). Old coordinates are replaced with new ones.

bk79, 2016-03-11:
What about the altitude? Does it changes it as well or still remains the same?

digitalentity, 2016-03-11:
Altitude is changed when you move throttle stick. When you only use
pitch/roll desired altitude is not changed.

bk79, 2016-03-11:
ok thanks for the explanation I thought that since GPS Coordinate are 3D I thought it was changing with the new position (if during the flight altitud changed as well)

closed 2016-03-11 by bk79