iNavFlight/inav #108: Board position alignment

Grahamtisse, 2016-03-11:
I have discovered an issue regarding the Board Alignment with the naze32.  No matter what degrees of Yaw Rotation is input, the axis remains unchanged!

digitalentity, 2016-03-11:
Please read the documentation. Board alignment is in deg*10 units.

Grahamtisse, 2016-03-11:
Thank you for a quick responce.  Im new to github and could you point me in the direction where this is.

oleost, 2016-03-11:
http://www.rcgroups.com/forums/showthread.php?t=2495732&page=1

https://github.com/iNavFlight/inav/wiki

And please try to dig around and search before asking...
Also use forums for support instead of github.

Grahamtisse, 2016-03-11:
Thank you for the advise.  Did not realise this kind of information is on
forums.

Best Regards

Graham

csurf, 2016-03-12:
@Grahamtisse please close out issue if it's been resolved... :)

closed 2016-03-12 by digitalentity