iNavFlight/inav #109: inav does not provide a clean API for GPS fix quality

stronnag, 2016-03-11:
GPS fix quality as reported by MSP or LTM is less precise than the data available in the FC. 

For MSP_RAW_GPS, the value of STATE(GPS_FIX) is returned. This does not match the MSP spec (0 or 1); the macro returns 2 (which a ground station might mis-construe as 2d-fix).

For LTM there is (mea culpa) an arbitrary calculation based on STATE(GPS_FIX) and the number of satellites.

For at least UBLOX (by far the most common protocol), we know the fix status (none, 2-d, 3-d). It would be useful if this was available as a global API, then we could abuse MSP_RAW_GPS with the real value (rather than 0 or 1), and provide an accurate indication in LTM. 

Finally, it could also be logged in blackbox.

This enhancement would improve the ability to debug GPS loss related issues.

digitalentity, 2016-03-12:
Maybe abuse the MSP_RAW_GPS with more types of fix? 0 - no fix, 1 - 2D fix, 2 - 3D fix? Fix type can be exposed in the `gpsSol` global and logged to blackbox as additional field in GPS frame.

digitalentity, 2016-03-12:
This change can be made along with #84

DzikuVx, 2016-03-12:
It would be very nice to be able to get HDOP in a simple way. Maybe a new CLI command, since with new MSP commands there is still a problem with waiting for it to be implemented in Configurator(s)

stronnag, 2016-03-12:
@digitalentity, yes having the fix type in the global would be excellent, then we report it as you suggested in RAW_GPS (and LTM GFRAME, and blackbox). This would be really good for live telemetry as as well as post mission debug.

digitalentity, 2016-03-16:
With #84 HDOP will be reported as first sat signal level:
[image]

@stronnag, please see #115

skaman82, 2016-03-16:
great!

digitalentity, 2016-03-16:
@skaman82 are you still working on Configurator chages?

skaman82, 2016-03-16:
Yes, I do. Had not enought time for that lately but I will work on some stuff with nighthawk32 in the next weeks. First we will add some new LED functionality like showing a GPS fix status.

skaman82, 2016-03-16:
If you need some help with the GUI I am happy to help out if I can

digitalentity, 2016-03-16:
@skaman82 be aware that iNav will report GPS fix type in MSP_RAW_GPS as following 0 - no fix, 1 - 2D fix, 2 - 3D fix. This will keep things compatible with exitsing tools that look for zero/not zero value while providing a more detailed fix info.

skaman82, 2016-03-16:
@digitalentity I haven't been following the changes for a couple weeks. I think if the 3D fix is reported differently than in original CF, it is very unlikely that it will make in the Configurator because it will not work with original CF. If you change the reporting, it should be merged first in original CF branch. Then it will be no problem to implement it into the GUI.

digitalentity, 2016-03-16:
@skaman82 with current implementation of Configurator does not distinguish between 2D fix and 3D fix, both will be reported as 3D fix by Configurator.
No fix will be reported correctly.

skaman82, 2016-03-16:
@digitalentity thanks. Yes that will work then! :)

skaman82, 2016-03-16:
@NightHawk32 FYI

digitalentity, 2016-03-16:
I'm trying to keep things as much compatible with original CF as I can, but this is not always possible. I'd be glad to see WP support in the Configurator but that's likely not going to happen until iNav is merged back to CF.

skaman82, 2016-03-16:
I am honestly not seeing iNAV merged to CF any time soon anymore ... But let's hope that I am wrong :)

skaman82, 2016-03-16:
What do you think of that?

No fix = red LED flashing
2D fix = yellow LED flashing
3D fix = green LED always on

Further discussion on LED function in here: https://github.com/skaman82/cleanflight-configurator/issues/67

digitalentity, 2016-03-16:
@skaman82 same here. I can see Cleanflight becoming a racing-only firmware and GPS support completely removed in favour of nifty racing features. I only hope @hydra will prove me wrong :smile:

digitalentity, 2016-03-16:
> Maybe abuse the MSP_RAW_GPS with more types of fix? 0 - no fix, 1 - 2D fix, 2 - 3D fix? Fix type can be exposed in the gpsSol global and logged to blackbox as additional field in GPS frame.

Done. MSP_RAW_GPS reportes 0/1/2 depending on a fix type. Fix type also available as gpsSol.fixType global. 
STATE(GPS_FIX) means 3D fix (gpsSol.fixType == 2)

Next step - add to LTM and blackbox.

stronnag, 2016-03-16:
Thanks. LTM uses 0 = no gps, 1=gps, no fix, 2=2D, 3=3D (take from KipK's original implementation), so consumers have to handle this slightly differently, but I don't see that can be avoided without breaking legacy.

stronnag, 2016-03-17:
Tested the new MSP_RAW_GPS and it is reporting fix type as expected. Also HDOP appears sane in the configurator (Neo 6M).

digitalentity, 2016-03-18:
Thanks! I'm closing this now. It's not very "clean" as in "well implemented and easy to use" but we can improve it later.

closed 2016-03-18 by digitalentity