iNavFlight/inav #115: LTM - Allow GPS HDOP to be reported via LTM

digitalentity, 2016-03-16:
We should probably think about modifying LTM to allow HDOP value to be sent to the groundstation/OSD. @stronnag, @oize what do you think?

DzikuVx, 2016-03-16:
I like this idea very much!

digitalentity, 2016-03-16:
We need to consider compatibility with existing tools. While I'd love to modify G-Frame and add another field to it, it will break compatibility with GhettoStation, EzGUI etc.

We have N-frame (navigation status) for reporting extra navigational data, maybe we should report HDOP there? Or should we add an extra frame for reporting HDOP and may be some other values?

stronnag, 2016-03-16:
I can't say I have a strong view, LTM is meant to be lightweight @2400 baud, but if we don't report too frequently, that should not be a problem. As there is no length field, extending any field (including N which is at least binary content compatible with MW/MSP) would NOT be my preference, rather there should be a new frame type.

DzikuVx, 2016-03-17:
Eventual new frame does not have to be sent very often. Additional data like HDOP or whatever else would be put there is not changing very often.
And extending frame N would break compatibility with frame N existing consumers

digitalentity, 2016-03-17:
Comments taken into consideration. There will be an additional frame (X - as in eXtented info), reported instead of one of N-frames, we won't be consuming more bandwidth.

What data should be there in the new X-frame?
1) HDOP
2) ???

DzikuVx, 2016-03-19:
My suggestion:
* gpsSol.hdop capped at 255 as uint8_t  since there is no need for more 
* gpsStats.lastMessageDt, the question is if we need max precision here? I would send it as millis / 10, cap at 255 (2.5s) and sent as uint8_t
* (gpsStats.errors / gpsStats.packetCount) * 1000 as uint16_t
* ???

stronnag, 2016-03-20:
I think HDOP needs to be uint16_t, I really don't care about the difference between 100 and 200 (*0.01), but I do about the difference between 255 and 9999. The raw data from the device is uint16_t, and it hardly seems worth diluting its precision for 1 byte.

digitalentity, 2016-03-20:
I tend to agree with @stronnag, HDOP is in x100 units and limited to 99.99, even if I keep one decimal point it's not possible to fit in 1 byte.

stronnag, 2016-03-26:
While we're discussing HDOP, the MSP_GPSSTATISTICS messages is broken (at least in testing), it should be 
````
        headSerialReply(20);
````
not 32.
(src/main/io/serial_msp.c#1144)

digitalentity, 2016-04-03:
@stronnag fixed in 0eb7139

stronnag, 2016-04-03:
Thanks, another local patch bites the dust:)

digitalentity, 2016-07-23:
At the moment I'm going to implement a new X-frame (sent at 1Hz rate) as following:
1) HDOP (uint16_t)
2) 4 bytes reserved for future use (set at zero at the moment)

This way frame will contain 6 bytes of data and it will be possible to extend it without modifying the frame size. @stronnag, what's your oppinion on this?

stronnag, 2016-07-23:
Wonderful. I no longer have to apply my private patch for similar functionality :-).

digitalentity, 2016-07-23:
How does your patch look like? Maybe we can reuse it here.

stronnag, 2016-07-23:
Similar, 1Hz but only when hdop changed. No reserved space  (as it was a private hack) --- having the reserved is a **good** idea.
````
+++ b/src/main/telemetry/ltm.c
@@ -88,6 +88,7 @@ static telemetryConfig_t *telemetryConfig;
 static bool ltmEnabled;
 static portSharing_e ltmPortSharing;
 static uint8_t ltm_crc;
+static uint16_t last_hdop;
 
 static void ltm_initialise_packet(uint8_t ltm_id)
 {
@@ -238,6 +239,14 @@ static void ltm_oframe()
     ltm_serialise_8(STATE(GPS_FIX_HOME) ? 1 : 0);
     ltm_finalise();
 }
+
+static void ltm_xframe()
+{
+    ltm_initialise_packet('X');
+    ltm_serialise_16(last_hdop);
+    ltm_finalise();
+}
+
 #endif
 
 #if defined(NAV)
@@ -288,7 +297,14 @@ static void process_ltm(void)
         ltm_gframe();
 
     if (current_schedule & LTM_BIT_OFRAME)
+    {
         ltm_oframe();
+        if (last_hdop != gpsSol.hdop)
+        {
+            last_hdop = gpsSol.hdop;
+            ltm_xframe();
+        }
+    }
 #endif
 
     if (current_schedule & LTM_BIT_SFRAME)
@@ -299,6 +315,9 @@ static void process_ltm(void)
         ltm_nframe();
 #endif
 
+
+
+
     ltm_scheduler = (ltm_scheduler + 1) % 10;
 }
 
@@ -329,6 +348,7 @@ void initLtmTelemetry(telemetryConfig_t *initialTelemetryConfig)
     telemetryConfig = initialTelemetryConfig;
     portConfig = findSerialPortConfig(FUNCTION_TELEMETRY_LTM);
     ltmPortSharing = determinePortSharing(portConfig, FUNCTION_TELEMETRY_LTM);
+    last_hdop = 0;
 }
 
 void configureLtmTelemetryPort(void)
````

digitalentity, 2016-07-23:
Thanks! I'll keep HDOP in first word of X-frame and add 4 reserved bytes after that to keep things compatible with mwp. I also like the idea of reporting X-frame only when hdop changes to save bandwidth.

PR #374 (2016-07-23): LTM X-frame to report HDOP
Closes #115
closed 2016-07-23 by digitalentity