iNavFlight/inav #116: Flight mode configuration and switching issues

digitalentity, 2016-03-17:
Reported by @stronnag:

> Something wierd is happening between at least 'testing' and the configurator. If I check the setting of the machine that successfully executed WP as http://www.rcgroups.com/forums/showp...postcount=4018, then the configurator shows NAV_WP is not set, and HOME_RESET is on my WP switch (and multiple other switches). I do not have HOME_RESET enabled.
> 
> Something very strange, as flipping the switch executed a WP mission on Monday and I've not changed any settings. Glad I didn't know that when I performed the mission test.
> 
> However, if I impose the same CLI settings on e.g. gps-remove-svinfo branch, then the NAV_WP switch is displayed correctly in the configurator; something not right in 'testing'?

digitalentity, 2016-03-17:
@stronnag 
Not really a bug. I added a new BOX mode - GCS NAV on request from @oize, CLI mode IDs got shifted. 1.0.2 will have a big warning to re-configure AUX settings.

Oize, 2016-03-17:
Could you please not metion me here. I don't even know you guys :)
Good luck in this project!

digitalentity, 2016-03-17:
@Oize, mea culpa, I ment @eziosoft (he goes by oize on RCG) :smile:

stronnag, 2016-03-17:
I think it is a bug. If I save the settings as:
[image]
Then I disconnect / reconnect (or reboot), then I get this:
[image]
(or worst, sometimes with HOME_RESET on multiple switches. 
This is after flashing with clear eeprom, and re-inputting the switches.

stronnag, 2016-03-17:
And, if I change the order of boxnames in the enum in `./src/main/io/rc_controls.h` to match that in which the names are defined in `./src/main/io/serial_msp.c` (i.e. move `BOXGCSNAV` to the end, then saving NAV WP works again.

And, yes, I distinctly recall hydra telling us that  `permanentId` fixed all this ...

digitalentity, 2016-03-18:
@stronnag 
I think I've spotted the bug. MSP_MODE_RANGES assumed that boxes[] array is ordered and indexed exactly as boxId_e enum, which may not be true. Saving modes worked correctly, retreiving them and sending to configurator that's what failed.

Fixed, thanks for reporting

commit f063c46e (2016-03-18): MSP: Fixed incorrect assumption about BOX ordering. Fixes #116
commit 8cd2ac2a (2016-03-18): MSP: Fixed incorrect assumption about BOX ordering. Fixes #116
closed 2016-03-18 by digitalentity