iNavFlight/inav #119: FW - Automatic cruise throttle estimation

digitalentity, 2016-03-18:
From RCG member:

> Just an idea here. For FW. May be multirrotors too.
> 
> For airplanes, the optimal, cruise speed,  is reached when you get the highest value of speed with the mininum amount of power. Id est, the minimal value for power/speed ratio. A filter for this should be enought to have a stable value of this ratio.
> 
> May be we could have a switch in gps navigation modes to accomplish this automagically. I guess we could have longer flying times and a good thing for emergency RTH. And  know much more about the performance of our models in FPV long flights.

sppnk, 2016-03-18:
great. I think ground speed is the variable we need for this future experiment, because the final goal is to get longer ranges in our flights.

As another idea, I am flying in small FPV planes w/o current sensor (cause reducing weight), but the amperage  data estimated throught throttle stick value in MWOSD  is good if well calibrated. Sure it changes if the day is windy or so, but for long distance fpv flight I guess everybody tried to have a clear low wind day. So may be this can be got just with voltage value...

digitalentity, 2016-03-18:
@sppnk 
We don't want the to do numbers here, there is direct (and mostly linear) relation with throttle and power. We'll just minimize speed/throttle ratio.

sppnk, 2016-03-18:
good. This could be fusionated with future CRUISE mode #92, with a parameter to switch on/oFF throught CLI

sppnk, 2016-07-25:
@digitalentity , as I have posted in other thread, the main gps navigation goals of inav have been reached. This 'power/speed or power/thr ratio to minimum'  will be very interesting to test. I dont expect big surprises here, but may be  we can extend the flight times a little. Just to think about. :-)

PS: may be this could be usefull for MR aswell, I dont know.

sppnk, 2016-08-16:
I have reading about how is managed this in other systems, and I guess I was wrong.

The only parameter we need to minimize for max range, is mAh/km, (or mA/speed), not Power/speed. The voltage has nothing to do when talking about 'fuel' per km. 

A big filter over the throttle cruise control putting an optimal speed would do the trick.

sppnk, 2016-08-25:
@digitalentity , you can close this, because I have opened #506 including the same, and #92 is still open.

closed 2016-09-01 by digitalentity