iNavFlight/inav #121: Default values for FP-PID and strange behavior

DzikuVx, 2016-03-19:
I know that controller is far from being ready, but today I took it for a fly in my 250 on 3S with Flip32:
* looptime 2000
* rates: 0.35 0.35 0.30
* build based on e181e7445dc970395aa485502160af5b1edfb051

Here are my notes on that experience:

1. default PIDs are flyable
1. quad had a tendency to drop the nose on tight turns and when finishing rolls and loops
1. Default PIDs caused visible oscillation above 70% of throttle when flying straight. What is intresting, oscillations were also appearing on tight turns on throttle lower than 70%
1. Lowering Roll/Pitch PIDs to 4/60/30 caused oscillations to go away, but tendency to drop the nose slightly increased
1. I think "nose dropping" is connected to new I term behavior. I think so because despite AirMode copter was unable to old attitude immediately after cutting throttle and immediately after punching throttle up to stop the fall.

Too bad I do not have BB Logs on that but I lost SD card last weekend somewhere on the airfield

digitalentity, 2016-03-20:
@DzikuVx 
Thanks for reporting. More D will help with oscillations. You may also want to increase I-term to fight the "nose drop" issue. 

Defaults are far from perfect. On my 250-sized quad I had to "up" P and D considerably (up to 9-10 on Pitch) to make it more stable. On my 450-sized quad I had to go as high as 7-7.5 on P (in Configurator).

Also, on asymmetric quads (ZMR-style) P should be much higher on Pitch than Roll.

DzikuVx, 2016-03-20:
@digitalentity 
If weather will allow (raining now :( ) I will give it another try with TPA and more I-term to see if "nose dropping" and oscillations would go away. With logs this time

digitalentity, 2016-03-20:
Great, thanks!
I think this is all up to proper tuning and good initial defaults. Don't be afraid to change P,I,D coefficients considerably - we don't really know yet what their values should be.

DzikuVx, 2016-03-20:
OK, I had another session with FP-PID today. I've bought new SD card, but it was crap. around 20% of frames lost with denom 2 (looptime 2000) so I will even do not post those. From the scraps of data I have, here are some conclusions:

1. Default PIDs caused oscillation on both roll and pitch (I see scraps of this on blackbox logs). From yesterdays session I know that lowering P to about 4 makes oscillations go away, but then quad becomes unpredictable
1. New D term computation and filtering is awesome. D stays sane and "smooth"
1. Either AirMode is not working or there is smth wrong with I term on cutting throttle and pushing it forward to stop the fall. Maybe something is preventing it to change fast enough? My quad's SG is far from perfect, moved back. I was unable to fix this even with I on Pitch moved to 160. Strange... Here is screenshot from BB log covering showing that: [image]
1. Some fighting with TPA resulted with TPA 40% with start at 1600.
1. Bottom line: I was able to get a tune that allowed me to fly 4 LiPos in total freestyle: rolls, tight turns, power loops, free falls. Oscillations were appearing from time to time, but only when pushing quad to do something that he was not willing to do. "Nose dropping" was also kicking in, but very rarely. It was not a good tune, but it was a tune allright. I was fighting with quad only from time to time.

At the end, I ended up with:

Rates: 0.35 / 0.35 / 0.22
Roll: P=4.1 I=120 D=100
Pitch: P: 6.1 I=160 D=100
TPA: 40% at 1600
Looptime: 2000

Maybe tomorrow I will try again with those values at start point. And maybe I will find SD card with good enough speed to write 1:1 at 2000 looptime

DzikuVx, 2016-03-20:
@digitalentity I see you ported new mixer. I can give it a try tomorrow without TPA and latest PIDs I was able to tune today

digitalentity, 2016-03-21:
I gave new mixer a short flight test today on my F450. It flies and it definitely feels more stable in flight at low throttle and full throttle.

DzikuVx, 2016-03-21:
OK, did some more tests. New mixer helped, but did not solved the problem of oscillations on high throttle. After some flying, I was able to come up to:

TPA: 20% at 1600 throttle
Roll: 3.5 / 160 / 80
Pitch 4.5 / 250 / 80

Quad reaction on adding and cutting the throttle was better comparing to yesterday. Also rolls were smoother. So there is a progress but I'm still far from being happy with results. 

I've created pull request #124 with changed Iterm prescaling and new defults altered accordingly

DzikuVx, 2016-04-03:
OK, I'm closing it

closed 2016-04-03 by DzikuVx

maciek01, 2016-04-11:
DzikuVz - what frame and FC you used with these PIDs? 250?

DzikuVx, 2016-04-11:
@maciek01 Yes, that was 250. My 600 allowed higher P but I stayed quite similar