iNavFlight/inav #123: iNav 1.0.1 ignores sero rate on CC3D

CopterFail, 2016-03-20:
Here is the effect, easy to reproduce:

Version: iNav 1.0.1 release
Hardware: CC3D
Mixer: fixed wing
Mode: Passthru
Servos: 3 + 4

Connect a receiver, select passthu and set all servo rates to 0%.
Change pitch and nothing will happen, as expected.
Change roll and the servos move with 100% rate.

I've tested the same with CF 1.12. release: works ok.

digitalentity, 2016-03-21:
Exactly the same setup: 1.0.1, CC3D, flying wing mixer, passthru more, servos 3+4 (4&5 on motor tab).
Rate 100% - servos move with both pitch and roll stick
Rate 0% - servos don't move at all.

digitalentity, 2016-03-21:
My guess would be misconfiguration. Serovs are numbered 0,1,2... at Servo tab and 1,2,3... at motor tab.

closed 2016-03-21 by digitalentity

CopterFail, 2016-03-21:
I know the index problem - to avoid this missunderstanding "ALL" servo rates were set to 0. 
I've stepped through the source. Servo output and mixer does fine. No problem with naze. I expect a side effect problem with the cc3d.

digitalentity, 2016-03-21:
@CopterFail 
No, still update to reproduce the issue on CC3D

CopterFail, 2016-03-22:
Resetting all the parameters after flashing removes the problem. Sorry - my fault.