iNavFlight/inav #130: HW - Support F4-based targets

digitalentity, 2016-03-24:
Some support for F4 is done by @sambas already, it has to be ported to iNav

Linjieqiang, 2016-03-24:
Great~! I'm going to design a flight control board for this firmware.The detailed list includes STM32F407,MPU9250,M8N GPS,MS5611~!

digitalentity, 2016-03-24:
@Linjieqiang amazing! 
My few cents regarding new FC:
 1. Please consider using MPU6500 as it doesn't include the magnetometer
 1. Place the gyro on SPI bus for faster communication
 1. You may want to use MS5607 baro - it's much cheaper and accurate enough
 1. If you are going to design a GPS module as well, please consider using NXP MAG3110 chip as HMC mag chips are EOL
 1. Don't include a voltage regulator onboard. It takes space and makes the FC less crash-proof
 1. Use level shifters (3.3V - 5V) for SONAR Echo and LEDSTRIP pins to make operation noise-proof.

Linjieqiang, 2016-03-24:
I suggest you should design a flight board and support the inav project perfectly fine like SPrcaingF3.The GPS module will be bought from Taobao.yeah,I will design for sonnar~!I will think about your suggestion.Thanks

brat002, 2016-03-25:
Please, add support of Sparky 2.0.

ScienceFaction, 2016-05-06:
Please add support for elle0 http://1bitsquared.com/products/elle0-autopilot
http://wiki.paparazziuav.org/wiki/Elle0
Its an F4 board with the sensors on an spi bus designed for paparazzi.
As paparazzi is open source getting the necessary porting info may be straight forward.

digitalentity, 2016-05-06:
@Linjieqiang @ScienceFaction @brat002 
I don't have any F4-based hardware to debug and test so this proibably won't happen any time soon. Sorry.

ScienceFaction, 2016-05-06:
@digitalentity How about if I get you an elle0?

digitalentity, 2016-05-06:
@ScienceFaction 
Then I'll get it working in INAV in a month or two (depending on how much time I have to spare). Drop me a message (my email address is in my GitHub profile).

martinbudden, 2016-07-09:
Closing, duplicate of issue https://github.com/iNavFlight/inav/issues/274

closed 2016-07-09 by martinbudden