iNavFlight/inav #131: FW - Speed-aware navigation

digitalentity, 2016-03-24:
Option to stabilise speed, not throttle

oleost, 2016-04-04:
Possible solution

http://fpvlab.com/forums/showthread.php?27854-NAZE32-for-fixed-wing&p=785877&viewfull=1#post785877

""The only thing the FW actually do is checking if ground speed is >5m/s.
If its lower it will increase the throttle to push forward against the wind.
But it's based cruise throttle as speed.
It never go below that in cruise only when descending.""

digitalentity, 2016-04-04:
@oleost looks like a good solution for airplanes that don't have pitot tube sensor

digitalentity, 2016-04-04:
@oleost 
Did a quick and dirty implementation of PatrickE's min speed guard - see "fw-throttle-controller" branch.

oleost, 2016-04-04:
@digitalentity https://vimeo.com/161521808

At 01:09 its below 5m/s. I think I heard the throttle go up, but not sure.

closed 2016-07-13 by digitalentity